fedora-iot/iot-distro #1: Osbuild created deliverables use the wrong naming format

paulwhalen, 2023-10-03:
**Describe the bug**
Osbuild created deliverables use the wrong naming format:

ISO   - Fedora-IoT-ostree-39.20231003.0-20231003.0.aarch64.iso
DISK - Fedora-IoT-39.20231003.0-20231003.0.aarch64.raw.xz

**Expected behavior**

These should follow the Fedora's image naming policy. 

Expected format: 
 - '%(release_short)s-%(disc_type)s-%(arch)s-%(version)s-%(date)s%(type_suffix)s.%(respin)s.iso'

Example- Fedora-IoT-ostree-aarch64-39-20231003.0.iso

juhp, 2024-01-23:
I came here to report this too: it kinda breaks my `dl-fedora` tool:
I added support for IoT but it can't list local iso because of this problem,
and I don't want to add a hack for it...

miabbott, 2024-01-26:
@achilleas-k it was noted that you said you would look into this; if this is not the case, let us know!

achilleas-k, 2024-01-26:
I do intend to look into it yes.  Was going to comment on this issue.  Feel free to assign it to me if you want to make it "official".

achilleas-k, 2024-01-26:
Someone who understands how Pungi and Koji interact better than I do will have to correct me but it seems to me that the format of the file is in fact:
```
%(release_short)s-%(disc_type)s-%(arch)s-%(version)s-%(date)s%(type_suffix)s.%(respin)s.iso
```
as mentioned in the first comment here.

The `version` component in the pungi config is set to `"!OSTREE_VERSION_FROM_LABEL_DATE_TYPE_RESPIN"` [1], which according to the pungi docs, becomes `$VERSION.$RELEASE` (e.g., `39.20240126.0`) [2].

I think this is confirmed by looking at any IoT build in Koji [3].
The Version here at the top is `39.20240122.0`, which I assume is the value of the `%(version)s` component in the filename.

I could be wrong though.

[1] https://pagure.io/fedora-iot/pungi-iot/blob/main/f/fedora-iot.conf#_216
[2] https://docs.pagure.org/pungi/configuration.html#ostree-settings
[3] https://koji.fedoraproject.org/koji/buildinfo?buildID=2366443

nullr0ute, 2024-01-31:
This PR resolved the issue! https://pagure.io/fedora-iot/pungi-iot/c/2448ed69c8dff6e57eddcea9f443851def85188d?branch=main

closed 2024-01-31 by nullr0ute

juhp, 2024-05-23:
I think it is better now but rawhide should be `Rawhide` not the fedora version to be consistent with other editions.

Compare:
```
~$ dl-fedora -nd rawhide silverblue
https://dl.fedoraproject.org/pub/fedora/linux/development/rawhide/Silverblue/x86_64/iso/Fedora-Silverblue-ostree-x86_64-Rawhide-20240523.n.0.iso
~$ dl-fedora -nd rawhide iot
File 6% downloaded
https://dl.fedoraproject.org/pub/alt/iot/rawhide/IoT/x86_64/iso/Fedora-IoT-ostree-41-20240522.0.x86_64.iso
```