fedora-iot/iot-distro #6: support `bootupd` for bootloader management/updates

miabbott, 2023-10-17:
**Is your feature request related to a problem? Please describe.**

When we update Fedora IoT via `rpm-ostree upgrade`, the bootloader binaries remains unchanged due the fact that the `ostree` model does not cover changes to the bootloader. Thus, users need to manually intervene to workaround this limitation and ensure the GRUB + UEFI shim firmware are up-to-date.

**Describe the solution you'd like**

Fedora IoT should be produced in a way that allows the use of `bootupd` to update the bootloader successfully.

**Describe alternatives you've considered**

Current alternative is a manual intervention by the use to copy the GRUB + UEFI shim from the ostree commit to the `/boot` partition.  See https://github.com/fedora-silverblue/issue-tracker/issues/120#issuecomment-1177515110

**Additional context**

Supporting `bootupd` requires that Fedora IoT is built using "unified-core" mode, which is not yet supported by osbuild.

See also: https://gitlab.com/CentOS/cloud/issue-tracker/-/issues/6

miabbott, 2024-01-26:
It should be possible to start using `bootupd` in IoT regardless of using unified-core mode to compose the ostree commit, but it is not clear how we go about enabling it.

Additional investigation will be required.

miabbott, 2024-01-26:
Since Fedora IoT has switched to using unified-core for composes (see #7 ), we should be able to pursue enabling `bootupd`

If I look at how it was done for Silverblue (https://pagure.io/workstation-ostree-config/pull-request/403), the changes to the compose look pretty straight-forward:  just the inclusion of the `bootupd` package and a `postprocess` script (https://pagure.io/workstation-ostree-config/blob/main/f/bootupd.yaml)

@paulwhalen is this something we can try in Rawhide? Do you think we need a Fedora Change for it?

paulwhalen, 2024-01-26:
> @paulwhalen is this something we can try in Rawhide? Do you think we need a Fedora Change for it?

We've missed the change deadline for Fedora 40, but I don't think that prevents us from enabling it.

miabbott, 2024-01-29:
Looks like the Silverblue enablement hit some snags - https://github.com/fedora-silverblue/issue-tracker/issues/530

paulwhalen, 2024-02-26:
Enabled in [Rawhide](https://pagure.io/fedora-iot/ostree/c/4ef41ffa05c6973590756ea4307d8899904e46b9?branch=main) and included in [Fedora-IoT-41-20240223.0](https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-41-20240223.0/)

PR for [f40](https://pagure.io/fedora-iot/ostree/pull-request/42)

paulwhalen, 2024-03-05:
Enabled in F40 composes since [Fedora-IoT-40-20240227.1](https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-40-20240227.1/)

PR to add in [osbuild/images](https://github.com/osbuild/images/pull/500).

pcdubs, 2024-05-22:
This was later [reverted](https://pagure.io/fedora-iot/ostree/c/dc36d6caf078e60eebbf22b1786e53535aee7a15?branch=f40) due to 

Still enabled in F41.

miabbott, 2024-08-28:
From the [bootupd docs about integrating with a distrobution](https://github.com/coreos/bootupd/blob/main/README-devel.md#integrating-bootupd-into-a-distributionos), we have the first part of the puzzle in place:

https://pagure.io/fedora-iot/ostree/c/4ef41ffa05c6973590756ea4307d8899904e46b9?branch=main

The second part of the puzzle will probably require some changes to the osbuild code that generates the IoT disk images.

miabbott, 2025-03-26:
I think we can consider this done...

The automatic bootloader update on boot was added to Rawhide/F42 via https://pagure.io/fedora-iot/ostree/pull-request/67

But that has since been replaced by a proper update service provided by the Fedora `bootupd` RPM - https://src.fedoraproject.org/rpms/rust-bootupd/pull-request/16

Nothing left to do?

travier, 2025-03-28:
👍🏻