fedora-iot/iot-distro #8: Selinux denial when using disk re-encryption with FDO in F38

7flying, 2023-10-19:
**Describe the bug**

Using disk re-encryption FDO features with Fedora 38 gets a selinux denial. We cannot use them.

**To Reproduce**

1. Generate a F38 simplified installer with fdo options, sample blueprint:
```
name = "fedora-si-fdo"
description = ""
version = "0.0.1"
packages = []
modules = []
groups = []
distro = ""

[customizations]
installation_device = "/dev/vda"

[[customizations.user]]
name = "admin"
password = "$6$vBo.9c8SeguWtjmu$8cj9HGn6nX6rPQvWh.pbdqaD.8FvLuIEToMOh9vHIQjjM.7PGZFWHYGxEO1dxuQ7ajjzzyuLI4EH.W6/ndXrV0"
groups = ["wheel"]
[customizations.fdo]
manufacturing_server_url = "http://192.168.122.180:8080"
diun_pub_key_insecure = "true"
```
2. Run the FDO infrastructure: `fdo-admin-tool aio --directory=./aio run`, the serviceinfo-api-server must have some `diskencryption_clevis` config, such as:
```
  diskencryption_clevis:
    - disk_label: /dev/vda3
      reencrypt: true
      binding:
        pin: tpm2
        config: '{}'
```

**Expected behavior**

I expect the disk to be re-encrypted.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**

Fedora 38

```
bash-5.2# rpm-ostree status -b
State: idle
BootedDeployment:
● fedora-iot:fedora/38/x86_64/iot
                  Version: 38 (2023-10-19T07:54:31Z)
                   Commit: e190257652791f6214518765f5e6ccee4969d67f9f5004adfb7401c101a291b1
```

**Additional context**
These are the logs:

```
ct 19 15:30:21 localhost.localdomain systemd[1]: Starting fdo-client-linuxapp.service - FDO client...
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.300Z INFO  fdo_client_linuxapp > Found device credential at FileSystemPath { path: "/boot/device-credentials", deactivation_method: None }
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.443Z INFO  fdo_client_linuxapp > Got TO2 addresses: ["http://192.168.122.180:8081", "http://fe80::97e2:1716:6aa8:88ba:8081"]
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.443Z INFO  fdo_client_linuxapp > Performing TO2 protocol, URL: "http://192.168.122.180:8081"
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.564Z INFO  fdo_client_linuxapp::serviceinfo > Initiating disk re-encryption, disk-label: /dev/vda3, pin: tpm2, config: {}, reencrypt: true
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain audit[1228]: AVC avc:  denied  { search } for  pid=1228 comm="pwmake" name="cracklib" dev="dm-1" ino=37010 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.777Z ERROR fdo_client_linuxapp              > ServiceInfo failed, error: Error processing returned serviceinfo
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]: Caused by:
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     0: Error executing clevis
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     1: Error executing disk encryption for disk label /dev/vda3
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     2: Error rebinding clevis
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     3: Error binding clevis
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     4: Failed to bind clevis: ExitStatus(unix_wait_status(256)), stdout: , stderr:
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        Error: Password generation failed - required entropy too low for settings
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        Unable to generate a new key
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        Error adding new binding to /dev/vda3
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:        
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:  2023-10-19T15:30:21.779Z ERROR fdo_client_linuxapp              > Error performing TO2 ownership protocol
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]: Caused by:
Oct 19 15:30:21 localhost.localdomain fdo-client-linuxapp[1116]:     Error performing the ServiceInfo roundtrips with TO2 address http://192.168.122.180:8081
```

7flying, 2023-10-19:
Same denials in F39 
[image]

paulwhalen, 2023-10-19:
Reproduced on Fedora 39:

Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:  2023-10-19T16:21:58.175Z INFO  fdo_client_linuxapp::serviceinfo > Initiating disk re-encryption, disk-label: /dev/vda3, pin: tpm2, config: {}, reencrypt: true
Oct 19 16:21:58 fedora-39-iot-custom audit[1488]: AVC avc:  denied  { search } for  pid=1488 comm="pwmake" name="cracklib" dev="dm-1" ino=164196 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 16:21:58 fedora-39-iot-custom audit[1488]: SYSCALL arch=c000003e syscall=257 success=no exit=-13 a0=ffffff9c a1=7ffd0d8e1000 a2=0 a3=0 items=0 ppid=1477 pid=1488 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="pwmake" exe="/usr/bin/pwmake" subj=system_u:>
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:  2023-10-19T16:21:58.256Z ERROR fdo_client_linuxapp              > ServiceInfo failed, error: Error processing returned serviceinfo
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]: Caused by:
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:     0: Error executing clevis
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:     1: Error executing disk encryption for disk label /dev/vda3
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:     2: Error rebinding clevis
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:     3: Error binding clevis
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:     4: Failed to bind clevis: ExitStatus(unix_wait_status(256)), stdout: , stderr:
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        /usr/share/cracklib/pw_dict.pwd.gz: Permission denied
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        Error: Password generation failed - required entropy too low for settings
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        Unable to generate a new key
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        Error adding new binding to /dev/vda3
Oct 19 16:21:58 fedora-39-iot-custom fdo-client-linuxapp[1232]:        
Oct 19 16:21:58 fedora-39-iot-custom kernel: audit: type=1400 audit(1697732518.253:194): avc:  denied  { search } for  pid=1488 comm="pwmake" name="cracklib" dev="dm-1" ino=164196 scontext=system_u:system_r:fdo_t:s0 tcontext=system_u:object_r:crack_db_t:s0 tclass=dir permissive=0
Oct 19 16:21:58 fedora-39-iot-custom kernel: audit: type=1300 audit(1697732518.253:194): arch=c000003e syscall=257 success=no exit=-13 a0=ffffff9c a1=7ffd0d8e1000 a2=0 a3=0 items=0 ppid=1477 pid=1488 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="pwmake" exe=">
Oct 19 16:21:58 fedora-39-iot-custom kernel: audit: type=1327 audit(1697732518.253:194): proctitle=70776D616B6500323536

paulwhalen, 2023-10-24:
Opened in [RHBZ](https://bugzilla.redhat.com/show_bug.cgi?id=2245935)

miabbott, 2024-01-26:
@7flying could you confirm if this problem is still happening? Please close the issue if it is resolved.

closed 2025-06-16 by pcdubs