fedora-iot/iot-distro #10: `parsec.service` fails to start after upgradeing to F39

miabbott, 2023-10-27:
**Describe the bug**
After upgrading to F39, the `parsec.service` fails to start

**To Reproduce**

1. Staring from a previous version of Fedora IoT, rebase to the F39 release (i.e. `rpm-ostree rebase fedora-iot:fedora/devel/x86_64/iot`)
2. Reboot and observe the journal

**Expected behavior**
`parsec.service` starts successfully

**OS version:**

```
$ rpm-ostree status -b
State: idle
BootedDeployment:
● fedora-iot:fedora/devel/x86_64/iot
                  Version: 39.20231026.0 (2023-10-26T12:27:40Z)
                   Commit: 0599c27fe88ed2aaeb8144c7b604aaa69e31a94cbc384c894e29b27a077bdb6a
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
```

**Additional context**

This is a carry-over from the F37 and F38 release, see https://bugzilla.redhat.com/show_bug.cgi?id=2170957

Failed service:

<details>

```
$ sudo journalctl -b -u parsec.service
[sudo] password for core: 
Oct 27 09:21:13 localhost systemd[1]: Starting parsec.service - Parsec Service...
Oct 27 09:21:13 localhost parsec[773]: Error: Permission denied (os error 13)
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:13 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Scheduled restart job, restart counter is at 1.
Oct 27 09:21:13 localhost systemd[1]: Starting parsec.service - Parsec Service...
Oct 27 09:21:13 localhost parsec[798]: Error: Permission denied (os error 13)
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:13 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Scheduled restart job, restart counter is at 2.
Oct 27 09:21:13 localhost systemd[1]: Starting parsec.service - Parsec Service...
Oct 27 09:21:13 localhost parsec[813]: Error: Permission denied (os error 13)
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 09:21:13 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:13 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Scheduled restart job, restart counter is at 3.
Oct 27 09:21:14 localhost systemd[1]: Starting parsec.service - Parsec Service...
Oct 27 09:21:14 localhost parsec[815]: Error: Permission denied (os error 13)
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:14 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Scheduled restart job, restart counter is at 4.
Oct 27 09:21:14 localhost systemd[1]: Starting parsec.service - Parsec Service...
Oct 27 09:21:14 localhost parsec[817]: Error: Permission denied (os error 13)
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:14 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Scheduled restart job, restart counter is at 5.
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Start request repeated too quickly.
Oct 27 09:21:14 localhost systemd[1]: parsec.service: Failed with result 'exit-code'.
Oct 27 09:21:14 localhost systemd[1]: Failed to start parsec.service - Parsec Service.
```

</details>

SELinux denial:

miabbott, 2024-01-26:
@nullr0ute reports that he is working on this in F39/F40 and should have a fix in the near future

Progress happening on the linked RHBZ

nullr0ute, 2024-03-10:
This update should fix the issue:
https://bodhi.fedoraproject.org/updates/FEDORA-2024-a223050257

nullr0ute, 2024-03-12:
We have confirmation this is now fixed :)

closed 2024-03-12 by nullr0ute