fedora-iot/iot-distro #11: qemu-system-x86_64 not able to boot Fedora IoT

say-paul, 2023-10-27:
```
qemu-system-x86_64 \
-machine q35 \
-M accel=hvf \
-m 4096 \
-cpu Broadwell \
-drive file=image.raw,format=raw,index=0,media=disk,if=virtio \
-drive if=pflash,format=raw,readonly=on,file=/usr/share/OVMF/OVMF_CODE.fd \
-drive if=pflash,format=raw,file=/usr/share/OVMF/OVMF_VARS.fd \
-device virtio-net-pci,netdev=n0,mac=FE:30:2E:2B:71:95 \
-netdev user,id=n0,net=10.0.2.0/24,hostfwd=tcp::8080-:8080,hostfwd=tcp::8082-:8082,hostfwd=tcp::2223-:22 \
-rtc base=localtime
```
worked for Rhel for edge but not with fedora IoT

miabbott, 2024-01-26:
@say-paul can you confirm this is still happening with F39 or F40 IoT artifacts?  If so, please attach more logs that show the error or problem.

say-paul, 2024-01-30:
I can get it working in my Fedora machine , somehow it was not working mac-machine which I dont have now.
so closing the ticket