fedora-iot/iot-distro #9: Fedora IoT 39 iso installation error: boot loader install failed

7flying, 2023-10-27:
**Describe the bug**

`boot loader install failed` while trying to install `Fedora-IoT-ostree-x86_64-39-20231026.0.iso` using GNOME Boxes.

**To Reproduce**

1. Download Fedora-IoT-ostree-x86_64-39-20231026.0.iso
2. Select the iso in Boxes
3. Add user, select disk (default), and after the installation starts the error will show up.

**Expected behavior**
Fedora is installed
**Screenshots**

[image]

**OS version:**

Not applicable.

**Additional context**
Add any other context about the problem here.

7flying, 2023-10-27:
If you select to ignore the error and continue with the installation, it installs the image, asks for a reboot and then it hangs in the following screen: 
[image]

miabbott, 2023-10-27:
You'll need to break into one of the  terminals during the initial install via `Ctrl+Alt+F2` and grab some logs from `/tmp`

Not sure exactly which logs have the error...

closed 2023-10-27 by paulwhalen