fedora-iot/iot-distro #20: Cockpit fails to load in browser with `PR_CONNECT_RESET_ERROR`

poperigby, 2024-01-10:
**Describe the bug**
Cockpit has started to fail with `PR_CONNECT_RESET_ERROR` when navigating to it in Firefox. When I try to navigate to the HTTP version of the site, it just redirects me to the HTTPS one and gives the same error. HTTP-Only Mode is disabled in Firefox.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Install Cockpit
2. Start it with `sudo systemctl start cockpit.service`
3. Navigate to http://<SERVER_IP>:9090 in browser

**Expected behavior**
Cockpit loads

**Screenshots**
[image]

**OS version:**

```
State: idle
AutomaticUpdates: stage; rpm-ostreed-automatic.timer: last run 15h ago
BootedDeployment:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20240105.0 (2024-01-05T14:26:16Z)
               BaseCommit: 5a9c39f47c18ef49633e131da4225229b2f727f89ef4d2fb5a7efdc98e408964
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
          LayeredPackages: borgmatic btrfs-progs cockpit cockpit-pcp cockpit-podman distrobox fish git ncurses smartmontools wireguard-tools
```

djach7, 2024-01-12:
Hello @poperigby, I've tried to reproduce this issue but I haven't been able to. Do you have any other steps to reproduce or logs you could share? I will say outside of the steps you listed I did open the cockpit.socket and add the service to my firewall, as found in the cockpit docs here: https://cockpit-project.org/running#fedora

poperigby, 2024-01-12:
I don't have any other reproduction steps, sadly. I should mention that besides having `cockpit` installed, I also have `cockpit-pcp`, and `cockpit-podman`. 

As for logs, this is all I have for `cockpit.service`:

```
Jan 10 09:46:29 haddock systemd[1]: Starting cockpit.service - Cockpit Web Service...
Jan 10 09:46:29 haddock systemd[1]: Started cockpit.service - Cockpit Web Service.
Jan 10 09:48:05 haddock systemd[1]: cockpit.service: Deactivated successfully.
Jan 10 09:50:33 haddock systemd[1]: Starting cockpit.service - Cockpit Web Service...
Jan 10 09:50:33 haddock systemd[1]: Started cockpit.service - Cockpit Web Service.
Jan 10 09:52:53 haddock systemd[1]: cockpit.service: Deactivated successfully.
Jan 12 12:55:59 haddock systemd[1]: Starting cockpit.service - Cockpit Web Service...
Jan 12 12:55:59 haddock systemd[1]: Started cockpit.service - Cockpit Web Service.
```

`cockpit.socket` is already enabled, and I have already have the firewall rules setup.

I also might mention this only recently started happening.

poperigby, 2024-01-15:
Hmm, I updated and it seems to work now. Sorry for the trouble, and thank you for your help.

closed 2024-01-15 by poperigby