fedora-iot/iot-distro #24: Osbuild should create network installer options during the compose process

paulwhalen, 2024-01-19:
**Describe the bug**

Since moving to osbuild for our installer, we no longer produce the network installer bits needed for things like PXE boot or HTTP boot. 

Example in Fedora 39:
https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-39-20240118.0/compose/IoT/x86_64/os/

Now in Fedora 40:
https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-40-20240118.0/compose/IoT/x86_64/

Included on the osbuild produced installer iso:

```bash
EFI
└── BOOT
    ├── BOOTAA64.EFI
    ├── fonts
    │   └── unicode.pf2
    ├── grubaa64.efi
    ├── grub.cfg
    └── mmaa64.efi
images
├── efiboot.img
├── install.img
└── pxeboot
    ├── initrd.img
    └── vmlinuz
```

Missing from F40 is the boot.iso. [F39 boot.iso](https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-39-20240118.0/compose/IoT/x86_64/os/images/boot.iso).

nullr0ute, 2024-01-19:
In the lorax world this was generated as part of the ISO install build process but there's no reason why it couldn't/shouldn't be a separate artifact.

paulwhalen, 2024-01-23:
https://github.com/osbuild/images/issues/394

mmartinv, 2024-02-05:
After discussing this with osbuild members it looks like we would need to create a new image type where we don't include the ostree commit and probably try to minimize the anaconda installer size (https://github.com/osbuild/images/issues/409). To do that we would need to provide a new stage for osbuild to allow the removal of files from the filesystem tree.

7flying, 2024-02-05:
@mmartinv please create a follow-up issue as the result of your investigation