fedora-iot/iot-distro #25: `pmlogger_daily.service` failing to start

poperigby, 2024-01-24:
**Describe the bug**
`pmlogger_daily.service` is failing:

```
Jan 24 15:19:27 haddock systemd[1]: Starting pmlogger_daily.service - Process archives...
Jan 24 15:19:27 haddock systemd[1]: Started pmlogger_daily.service - Process archives.
Jan 24 15:19:28 haddock systemd[1]: pmlogger_daily.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 15:19:28 haddock systemd[1]: pmlogger_daily.service: Failed with result 'exit-code'.
```

I tried running the command that it runs `sudo /usr/libexec/pcp/bin/pmlogger_daily -E`, and that failed with an exit code of 1.

**To Reproduce**
1. Run `sudo systemctl start pmlogger_daily.service` or `sudo /usr/libexec/pcp/bin/pmlogger_daily -E`

**Expected behavior**
`pmlogger_daily.service` doesn't fail

**OS version:**

```
State: idle
AutomaticUpdates: stage; rpm-ostreed-automatic.timer: no runs since boot
BootedDeployment:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20240122.0 (2024-01-22T15:18:27Z)
               BaseCommit: 23398f5f1fc27cc8e6e56f967420eeb142f3295d4999251a1cc6f223497ef509
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
          LayeredPackages: borgmatic btrfs-progs cockpit cockpit-pcp cockpit-podman distrobox fish git ncurses smartmontools wireguard-tools

```

djach7, 2024-01-26:
Hey @poperigby, I tried to recreate this a little bit ago and unfortunately no luck. Here's my `rpm-ostree status`: 

```State: idle
Deployments:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20240122.0 (2024-01-22T15:18:27Z)
               BaseCommit: 23398f5f1fc27cc8e6e56f967420eeb142f3295d4999251a1cc6f223497ef509
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
          LayeredPackages: borgmatic btrfs-progs cockpit cockpit-pcp cockpit-podman distrobox fish git ncurses
                           smartmontools wireguard-tools
```

Here is what I get after running the service: 
```
Jan 26 10:07:07 localhost.localdomain systemd[1]: Starting pmlogger_daily.service - Process archives...
Jan 26 10:07:07 localhost.localdomain systemd[1]: Started pmlogger_daily.service - Process archives.
Jan 26 10:07:08 localhost.localdomain systemd[1]: pmlogger_daily.service: Deactivated successfully.
```

Do you have anything else in the journalctl logs that might be helpful? Or any other reproduction steps?

poperigby, 2024-01-30:
@djach7 No, unfortunately :confused:. This is the only thing it logged:

```
Jan 29 19:15:19 haddock systemd[1]: Starting pmlogger_daily.service - Process archives...
Jan 29 19:15:19 haddock systemd[1]: Started pmlogger_daily.service - Process archives.
Jan 29 19:15:19 haddock systemd[1]: pmlogger_daily.service: Main process exited, code=exited, status=1/FAILURE
Jan 29 19:15:19 haddock systemd[1]: pmlogger_daily.service: Failed with result 'exit-code'.
```

Do you know any way to make the output more verbose, or what this thing even does?

djach7, 2024-01-30:
So I've got a possible solution. This forum thread (https://forums.fedoraforum.org/showthread.php?331674-pmlogger-not-starting-after-Fedora-IoT-upgrade-to-39) implies that it might be a pcp user issue, try that and see if it helps? 

Also another question, are you running fedora 39 out of the box or have you upgraded from 38?

poperigby, 2024-01-30:
Changing the ownership of the directory to `pcp:pcp` fixed it! Yes, I did upgrade from 38. Is there any reason this wasn't done manually?

djach7, 2024-01-30:
Glad to hear it! I can't say I know why it wasn't done manually but I'll look into it for you. Is there anything else or is this issue ok to be closed?

poperigby, 2024-01-31:
> I can't say I know why it wasn't done manually but I'll look into it for you

Thank you :slightly_smiling_face: 

> Is there anything else or is this issue ok to be closed?

Nope! Thanks for your help!

closed 2024-01-31 by poperigby

nullr0ute, 2024-01-31:
> Changing the ownership of the directory to `pcp:pcp` fixed it! Yes, I did upgrade from 38. Is there any reason this wasn't done manually?

I presume you mean automatically, that would be a question for the pcp maintainers I believe, so out of scope for here.