fedora-iot/iot-distro #26: `fish` package on IoT doesn't have `ncurses` as a dependency, causing `clear` to fail

poperigby, 2024-01-25:
**Describe the bug**
As described above, installing `fish` doesn't pull in `ncurses` as a dependency, which causes the `clear` command to fail. This only seems to be the case on IoT, and not Kinoite, unless `ncurses` is installed by default on Kinoite so I didn't notice.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Run `rpm-ostree install fish`
2. Run `clear`

**Expected behavior**
Running `clear` run successfully, and clears the screen.

**OS version:**

```
State: idle
AutomaticUpdates: stage; rpm-ostreed-automatic.timer: last run 20h ago
BootedDeployment:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20240122.0 (2024-01-22T15:18:27Z)
               BaseCommit: 23398f5f1fc27cc8e6e56f967420eeb142f3295d4999251a1cc6f223497ef509
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
          LayeredPackages: borgmatic btrfs-progs cockpit cockpit-pcp cockpit-podman distrobox fish git ncurses smartmontools wireguard-tools

```

nullr0ute, 2024-01-26:
This is a dependency problem with the fish package itself, could you report it to the fish component first? A list of open bugs for it can be found here: https://bugzilla.redhat.com/buglist.cgi?bug_status=NEW&bug_status=ASSIGNED&classification=Fedora&component=fish&list_id=13404920&product=Fedora&product=Fedora%20EPEL

poperigby, 2024-01-30:
Yes, of course. Thank you for directing me to the right spot.

closed 2024-01-30 by poperigby

poperigby, 2024-01-30:
Huh, looks like the issue is actually gone now. Cool.