fedora-iot/iot-distro #28: Document how to use PXE boot

7flying, 2024-02-13:
Based on #24 document how to use PXE boot: Note the missing `os` directory in [Fedora 40](https://kojipkgs.fedoraproject.org/compose/iot/latest-Fedora-IoT-40/compose/IoT/x86_64/) vs [Fedora 39](https://kojipkgs.fedoraproject.org/compose/iot/latest-Fedora-IoT-39/compose/IoT/x86_64/).

(split from #27 )

**Goal**

- Document how to use PXE boot

**Acceptance Criteria**
- Docs are updated
- Add a use case that tests that PXE boot works

miabbott, 2024-02-13:
In the context of #24 and #27, the internal discussion lead us to the decision to document how to extract the PXE artifacts from the Anaconda ISO.

However, if PXE support is not highly desired by the community, this may not be required. If folks from the community have a need for PXE artifacts and can't migrate to HTTP boot, we'd like to hear from you.

pcdubs, 2024-05-22:
We've had no response or inquiry around PXE since this issue was opened. 

Before we close it out, we should remove any Release Criteria for `IoT ` that mentions a requirement that PXE works, also the testcase in the wiki matrix.

coremodule, 2024-09-29:
Removed the PXE boot testcase from the IoT test matrix. I also scoured the release criteria and docs, but found no other references to PXE boot. I do not have permissions to close this issue, @pcdubs could you mark this as closed?

pcdubs, 2024-10-02:
Thanks @coremodule ! Closing now