fedora-iot/iot-distro #31: Support for ROS2 on Fedora IoT with Robotics SIG

nullr0ute, 2024-02-16:
The [Fedora Robotics SIG](https://fedoraproject.org/wiki/SIGs/Robotics) is quite active and are working to package up [ROS2](https://docs.ros.org/en/rolling/index.html) into Fedora and are interested in how they could work with us to support it on Fedora IoT. It's also a target in EPEL for the enterprise use case as well as it's used in factory automation and all sorts of industrial robotics, not just fun cars :)

nullr0ute, 2024-02-16:
Some of the questions we have here is:
* Do we support it in a container or a ostree layer
* How do we support the hardware access
* How do we deal with media and things like complex cameras often used in these use cases
* Access and support for things like AI HW (GPU, NPU, FPGA etc)

miabbott, 2024-02-16:
@say-paul has shown interest in enabling ROS2 for IoT; he might be able to answer some of these questions.

Additionally @odra or @lukewarmtemp may be able to help with some of these questions

say-paul, 2024-04-17:
So with centos-bootc I am able to layer ROS2 and able create an installer out of it.: [discussion(https://discussion.fedoraproject.org/t/2024-03-28-fedora-robotics-sig-meeting/110059).  
looking forward to fedora-bootc image to do that on fedora.

say-paul, 2025-03-10:
Do we support it in a container or an ostree layer?

We will provide containers that can be built into images using bib.
We have robotics spin of bootc[images](https://gitlab.com/fedora/sigs/robotics/images), which builds ROS2 from source.  later, it would be good to have the ROS2 rpms built for fedora, we had discussion with open robotics , they can help us build all the packages if provided with the build systems.

How do we support hardware access?

1. Testing (HIL) -  using [jumstarter](https://jumpstarter.dev) to test applications in hardware will be physically connected the RPI with system; we can develop a jumpestarter driver to do it over wifi.
2. Deploying - flashing SD card though arm-image-installer 

How do we deal with media and things like complex cameras often used in these use cases

Access and support for things like AI HW (GPU, NPU, FPGA etc);

I think we need the communication protocols supported from the bootloader first

say-paul, 2026-10-12:
Fedora ROS2 testing repo is now available in ROS2 official [documemtation](https://docs.ros.org/en/rolling/Get-Started/Installation/RHEL-Install-RPMs.html)
for any issue  report to [fedora robotics SIG](https://gitlab.com/fedora/sigs/robotics/fedora-robotics-sig)

closed 2026-10-12 by say-paul