fedora-iot/iot-distro #32: F39 legacy installer is failing to complete in OpenQA

miabbott, 2024-03-06:
Reported by @adamwill on the IoT channel:

> btw the last few f39 IoT nightlies do not install with an error Multiple specifications found for remote "fedora-iot"
> https://openqa.fedoraproject.org/tests/2454758#step/_do_install_and_reboot/115
> same on x86_64 and aarch64
> has been happening since Fedora-IoT-39-20240226.0

pcdubs, 2024-03-06:
This happened because we pulled in the `fedora-iot-config` package to fix other breakage with the remote in F39 on the disk images. OpenQA is testing the legacy installer which sets the remote in the `fedora-lorax-templates` as well, so it conflicts. 

There are some, mostly cosmetic issues with the osbuild produced installer, could we move OpenQA over to test that In F39?

AdamWill, 2024-03-06:
It's possible, yes (though as I guess you noticed, it would be easier with https://pagure.io/fedora-iot/pungi-iot/pull-request/33 merged).

Ideally, what openQA tests should be what we actually deliver to people for use. Is there some 'official' distribution channel for updated IoT nightlies for stable releases? If so, which of the images is that delivering?

pcdubs, 2024-03-06:
To date we have not yet delivered an updated release, we were readying the first when the issue with the remote was discovered. When we do, we would need to use the Image Builder isos because of this conflict.

AdamWill, 2024-03-07:
okay, then I can flip openQA to testing those after the metadata is hopefully fixed in the next nightly.

pcdubs, 2024-05-22:
F39 is now EOL and IoT no longer produces the legacy installer. Closing.

closed 2024-05-22 by pcdubs

AdamWill, 2024-05-22:
Er, F39 is not EOL. F38 is the release that just went EOL.

pcdubs, 2024-05-22:
> Er, F39 is not EOL. F38 is the release that just went EOL. still, it seems we're not building F39 IoT installer images any more, so it's not much of a live issue any more.

In context of IoT it effectively is. We no longer build F39, anyone previously using stable is moved to f40.