fedora-iot/iot-distro #33: Create Fedora 40 Beta IoT candidates

pcdubs, 2024-03-13:
**Describe the bug**
Create the Fedora 40 Beta IoT compose, ensuring all blockers and Freeze Exceptions have been pulled in. 

Upstream request with blocker and freeze exceptions:
https://pagure.io/releng/issue/12007

nullr0ute, 2024-03-13:
Just documenting this here we need `koji list-tagged --latest f40-compose` tagged into f40-iot. Just a note to make sure we have this detail in the SOP

pcdubs, 2024-03-14:
Compose `Fedora-IoT-40-20240313.2` has all the needed blockers/FE's but the bootc image failed. 

Today's compose was complete - `Fedora-IoT-40-20240314.0`.

pcdubs, 2024-03-19:
Latest F40 IoT beta compose - https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-40-20240319.2/

SOP for Beta composes - https://github.com/fedora-iot/iot-docs/pull/59

pcdubs, 2024-03-21:
Latest beta, includes all RC10 blockers/FE's
 https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-40-20240321.0/

closed 2024-03-21 by pcdubs