fedora-iot/iot-distro #35: IoT 40/41 Fails to boot on x86 SBC

fayalalebrun, 2024-03-19:
**Describe the bug**
When attempting to boot after rebasing to version 40/41, no volumes can be found by dracut and the system fails to boot.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Install Fedora IoT 39 (In this case to a GHF51 SBC)
2. Rebase to Fedora IoT 40 or 41
3. Restart

**Expected behavior**
Fedora IoT boots successfully.

**Screenshots**
[image]

**OS version:**

```
  fedora-iot:fedora/rawhide/x86_64/iot
                  Version: 41.20240315.0 (2024-03-15T16:10:39Z)
                   Commit: ae3236721a484ff13d39ce5cad7f9e934b307e3016848747f7596031e5c3cba5
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
                     Diff: 348 upgraded, 10 removed, 13 added

● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20231103.0 (2023-11-03T15:32:03Z)
                   Commit: 257e2c9d3fe02d977859f01a4e8187181e7cf24d4d755fd89a06beb7c78bec61
             GPGSignature: Valid signature by 6A51BBABBA3D5467B6171221809A8D7CEB10B464
```

nullr0ute, 2024-03-19:
So it works in F-39 but not later releases? Can you provide details of the storage used, any special setup? Does it run vanilla Fedora 40 OK?

nullr0ute, 2024-03-19:
>                   Version: 39.20231103.0 (2023-11-03T15:32:03Z)

Is this the latest F-39 compose you tested? What kernel is that running?

fayalalebrun, 2024-03-19:
Correct, Fedora IoT 36, 37, 38, and 39 have all worked with this SBC. 40 or later has issues.

One odd thing I noticed was that `lsblk` shows no devices in the recovery shell. Perhaps it is missing a driver for the flash memory? 

Storage is LVM.
```
NAME                 MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk0              179:0    0 59.2G  0 disk
├─mmcblk0p1          179:1    0  600M  0 part /boot/efi
├─mmcblk0p2          179:2    0    1G  0 part /boot
└─mmcblk0p3          179:3    0 57.6G  0 part
  ├─fedora--iot-root 253:0    0 38.7G  0 lvm  /var
  │                                           /sysroot/ostree/deploy/fedora-iot/var
  │                                           /usr
  │                                           /etc
  │                                           /
  │                                           /sysroot
  └─fedora--iot-home 253:1    0 18.9G  0 lvm  /var/home
mmcblk0boot0         179:8    0    4M  1 disk
mmcblk0boot1         179:16   0    4M  1 disk
zram0                252:0    0  3.3G  0 disk [SWAP]
```

I update to the latest fedora 39 and rebased from there but saw no change:

```
$ uname -r
6.7.7-200.fc39.x86_64
```

```
Deployments:
  fedora-iot:fedora/devel/x86_64/iot
                  Version: 40.20240318.2 (2024-03-18T17:03:17Z)
                   Commit: d2cb49b07ab084a0a8b2a778f433623652c3de3f5639788707c7e6970c1ab40d
             GPGSignature: Valid signature by 115DF9AEF857853EE8445D0A0727707EA15B79CC
                     Diff: 341 upgraded, 11 downgraded, 14 removed, 4 added

● fedora-iot:fedora/stable/x86_64/iot
                  Version: 39.20240312.0 (2024-03-12T17:45:27Z)
                   Commit: 9c62419d6f8a8947edf6bb3f1a69c31235ac43c43bcd20f2141e58f050068dd2
             GPGSignature: Valid signature by E8F23996F23218640CB44CBE75CF5AC418B8E74C
```

I will have to check on Fedora 40.

nullr0ute, 2024-03-19:
> Storage is LVM.
> 
> ```
> NAME                 MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
> mmcblk0              179:0    0 59.2G  0 disk
 
The storage is mmc, LVM is a volume manager on top of the base storage.

On the f39 which works can you provide the output of `lsmod |grep 'mmc\|sdhci'` which should show what mmc driver the AMD platform uses. Also on F-40 can you see if there's any errors around mmc with `sudo dmesg |grep 'mmc\|sdhci'`

fayalalebrun, 2024-03-19:
Looks like a kernel bug. Feel free to close this issue.

Fedora 39:
```
[    6.028117] sdhci: Secure Digital Host Controller Interface driver
[    6.028991] sdhci: Copyright(c) Pierre Ossman
[    6.071287] mmc0: SDHCI controller on ACPI [AMDI0040:00] using ADMA
[    6.178797] mmc0: new HS400 MMC card at address 0001
[    6.191585] mmcblk0: mmc0:0001 DG4064 59.2 GiB
[    6.204113]  mmcblk0: p1 p2 p3
[    6.224709] mmcblk0boot0: mmc0:0001 DG4064 4.00 MiB
[    6.234367] mmcblk0boot1: mmc0:0001 DG4064 4.00 MiB
[    6.239270] mmcblk0rpmb: mmc0:0001 DG4064 4.00 MiB, chardev (239:0)
[   11.007524] EXT4-fs (mmcblk0p2): mounted filesystem b65c5073-a35f-49d7-9ba4-46a103de7750 r/w with ordered data mode. Quota mode: none.
```

Fedora 40:
```
[   10.597267] sdhci: Secure Digital Host Controller Interface driver
[   10.597272] sdhci: Copyright(c) Pierre Ossman
[   10.626391] genirq: Flags mismatch irq 7. 00000088 (mmc0) vs. 00002088 (pinctrl_amd)
[   10.627055]  ? __pfx_sdhci_thread_irq+0x10/0x10 [sdhci]
[   10.627082]  __sdhci_add_host+0x108/0x360 [sdhci]
[   10.627107]  sdhci_acpi_probe+0x3a8/0x500 [sdhci_acpi]
[   10.629180] mmc0: Failed to request IRQ 7: -16
[   10.630274] sdhci-acpi: probe of AMDI0040:00 failed with error -16
```

nullr0ute, 2024-03-19:
Yes, it's kernel issue, for reference the driver is `sdhci_acpi`

closed 2024-03-19 by nullr0ute

nullr0ute, 2024-04-22:
For reference looks like the fix landed in v6.9-rc2 and v6.8.3