fedora-iot/iot-distro #37: Rawhide deployment does not boot on Raspberry Pi 4 Model B Rev 1.4 8 GB

hrismarin, 2024-03-23:
**Describe the bug**
When I try to boot a Rawhide deployment (41.20240315.0) on a Raspberry Pi 4 Model B Rev 1.4 8 GB, the device becomes unresponsive.

**To Reproduce**
1. Flash Fedora IoT 39 or 40 raw image to microSD card with `arm-image-installer`.
2. `rpm-ostree rebase fedora-iot:fedora/rawhide/aarch64/iot`

**Expected behavior**
The `41.20240315.0 (2024-03-15T16:11:04Z)` deployment should boot.

**Screen output after the bootloader**
```
  Booting `Fedora Linux 41.20240315.0 (IoT Edition Prerelease) (ostree:0)'
EFI stub: Deconpressing Linux Kernel...
EFT stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```
Then the device freezes.

**OS version:**
The output of the `rpm-ostree status` command after rebase and before reboot.
```
[root@localhost ~]# rpm-ostree rebase fedora-iot:fedora/rawhide/aarch64/iot
⠠ Receiving objects; 99% (24951/24961) 1.2 MB/s 686.7 MB
3499 metadata, 21465 content objects fetched; 671510 KiB transferred in 557 seconds; 1.8 GB content written
Receiving objects; 99% (24951/24961) 1.2 MB/s 686.7 MB... done
Staging deployment... done

...

Changes queued for next boot. Run "systemctl reboot" to start a reboot
[root@localhost ~]# rpm-ostree status 
State: idle
Deployments:
  fedora-iot:fedora/rawhide/aarch64/iot
                  Version: 41.20240315.0 (2024-03-15T16:11:04Z)
                   Commit: d2c8b1f04509fa610e6a70adf4a915ca4297a05358588ccf10fb1cda5679c362
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
                     Diff: 356 upgraded, 11 removed, 13 added

● fedora-iot:fedora/stable/aarch64/iot
                  Version: 39.20231103.1 (2023-11-03T18:17:43Z)
                   Commit: cc8d419f72d84ac24d0a95e235c0cdf72844f73d9d6f42a41fcddf23dfb34f7d
             GPGSignature: Valid signature by 6A51BBABBA3D5467B6171221809A8D7CEB10B464
[root@localhost ~]# reboot 
[root@localhost ~]# Connection to 192.168.30.14 closed by remote host.
Connection to 192.168.30.14 closed.
```

**Additional context**
```
[root@localhost ~]# hostnamectl status 
...
     Hardware Vendor: raspberrypi
      Hardware Model: Raspberry Pi 4 Model B Rev 1.4
...
```

miabbott, 2024-03-25:
@nullr0ute I feel like you know the right questions to ask about this problem

pcdubs, 2024-03-25:
Could be - https://github.com/coreos/fedora-coreos-tracker/issues/1647

pcdubs, 2024-03-25:
Confirmed:

```
[root@rpi4-4 ~]# cat /proc/cmdline 
BOOT_IMAGE=(hd0,msdos2)/ostree/fedora-iot-07992683c92cd700013d7a63e6042e66a5a472c52d20df873391fa90a96dfdfb/vmlinuz-6.9.0-0.rc0.fc41.aarch64 modprobe.blacklist=vc4 rw coreos.no_persist_ip ignition.platform.id=metal root=LABEL=root ostree=/ostree/boot.1/fedora-iot/07992683c92cd700013d7a63e6042e66a5a472c52d20df873391fa90a96dfdfb/0
[root@rpi4-4 ~]# rpm-ostree status 
State: idle
Deployments:
  fedora-iot:fedora/devel/aarch64/iot
                  Version: 40.20240321.0 (2024-03-21T11:24:54Z)
                   Commit: a14a7b61a3c700c346ae9cd536f714fbbc6d26065a22e2afb7f215abf7a1d5a0
             GPGSignature: Valid signature by 115DF9AEF857853EE8445D0A0727707EA15B79CC
                     Diff: 7 upgraded, 137 downgraded, 3 removed, 1 added

● fedora-iot:fedora/rawhide/aarch64/iot
                  Version: 41.20240315.0 (2024-03-15T16:11:04Z)
                   Commit: d2c8b1f04509fa610e6a70adf4a915ca4297a05358588ccf10fb1cda5679c362
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
[root@rpi4-4 ~]# uname -r
6.9.0-0.rc0.20240314git480e035fc4c7.5.fc41.aarch64

```

nullr0ute, 2024-04-15:
So this is apparently fixed for the CoreOS use case, can we have confirmation it's fixed for IoT too please?

hrismarin, 2024-04-16:
I can confirm that it is fixed for IoT:
```
Boot Status is GREEN - Health Check SUCCESS
Last login: Tue Apr 16 16:06:42 2024 from 172.16.20.12
[root@localhost ~]# rpm-ostree status 
State: idle
Deployments:
● fedora-iot:fedora/rawhide/aarch64/iot
                  Version: 41.20240415.1 (2024-04-15T16:43:50Z)
               BaseCommit: 71ca901bf1c5d59980d935583a8dca088278f7a9851f6c3401479e27ff177d97
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
          LayeredPackages: gcc git koji libtirpc-devel make python3-fedora

  fedora-iot:fedora/stable/aarch64/iot
                  Version: 39.20231103.1 (2023-11-03T18:17:43Z)
               BaseCommit: cc8d419f72d84ac24d0a95e235c0cdf72844f73d9d6f42a41fcddf23dfb34f7d
             GPGSignature: Valid signature by 6A51BBABBA3D5467B6171221809A8D7CEB10B464
          LayeredPackages: gcc git koji libtirpc-devel make python3-fedora
```

closed 2024-04-16 by hrismarin