fedora-iot/iot-distro #38: document how to build, install, configure the IoT bootable container

miabbott, 2024-04-03:
As part of Fedora 40, Fedora IoT will have another means of building, installing, configuring a system via bootable containers.

https://fedoraproject.org/wiki/Changes/Fedora_IoT_Bootable_Container

As we roll out this new addition, we need to document how to use it:

- how can we customize the container?
- how can we create disk images from the container?
- how can users update their IoT systems using a container?
- etc

adelton, 2024-07-01:
The https://fedoraproject.org/wiki/Changes/Fedora_IoT_Bootable_Container says

Targeted release: [Fedora Linux 40](https://docs.fedoraproject.org/en-US/releases/f40/)

With Fedora 40 GA released, is any part of that change present? Or does it effectively slip to 40+ and the change page should be updated accordingly?

miabbott, 2024-07-01:
I think we are in an in-between state...there are Fedora `bootc` base images that are being produced (i.e. quay.io/fedora/fedora-bootc:40 ) but I don't believe we are building an IoT specific `bootc` image yet.

The roadmap issue (#53) contains an item "[Fedora IoT derived bootc image is defined and composed automatically](https://github.com/fedora-iot/iot-distro/issues/19)" which would close that gap.

miabbott, 2024-07-01:
@nullr0ute do you have any better info here?

nullr0ute, 2024-11-27:
We have bootc IoT specific containers as part of F-41, the container is fedora-iot:latest in the usual Fedora repos, there's an incomplete docs PR outstanding

closed 2024-11-27 by nullr0ute