fedora-iot/iot-distro #40: FDO client application is not enabled after installing the device with the simplified installer

mmartinv, 2024-04-12:
**Describe the bug**
After installing Fedora IOT 40 beta by using the Simplified Installer and passing the FDO manufacturing configuration, the installation finishes correctly, the manufacturing process takes place and the device is powered off.

However, in the first boot the device does not try to perform the FIDO onboarding protocol as expected.
Taking a deeper look it looks like the systemd service is disabled:
```console
systemctl list-unit-files fdo-client-linuxapp.service
UNIT FILE                   STATE    PRESET  
fdo-client-linuxapp.service disabled disabled

1 unit files listed.
```

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Install Fedora IoT 40 Beta:
```console
virt-install --connect qemu:///system \
             --name fedora-iot-simplified-installer \
             --os-variant fedora-rawhide \
             --boot uefi,loader.secure=false \
             --noautoconsole \
             --vcpus 1 --memory 3072 \
             --network network=default,model=virtio \
             --disk pool=default,size=30 \
             --cdrom Fedora-IoT-provisioner-40-20240319.2.x86_64.iso
```
2.  Reclaim and provision the system from https://provision.fedoraproject.org/portal/devices/
3. Connect to the system and check the FDO client application:
```console
systemctl status fdo-client-linuxapp.service 
○ fdo-client-linuxapp.service - FDO client
     Loaded: loaded (/usr/lib/systemd/system/fdo-client-linuxapp.service; disabled; preset: disabled)
     Drop-In: /usr/lib/systemd/system/service.d
              └─10-timeout-abort.conf
     Active: inactive (dead)
```

**Expected behavior**
The onboarding process is triggered in firstboot after the installation.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**
Fedora-IoT-provisioner-40-20240407.0.x86_64.iso

```console
rpm-ostree status -b
State: idle
BootedDeployment:
● fedora-iot:fedora/devel/x86_64/iot
                  Version: 40.20240319.2 (2024-03-19T17:58:04Z)
                   Commit: 4761cfc7d5c969b7a8c2b36d5c9980424af0a0127358b48dcb775857ec587406
             GPGSignature: Valid signature by 115DF9AEF857853EE8445D0A0727707EA15B79CC
```

pcdubs, 2024-04-12:
PR's to fix:
https://src.fedoraproject.org/rpms/fedora-release/pull-request/329
https://src.fedoraproject.org/rpms/fedora-release/pull-request/330

Bugzilla:
https://bugzilla.redhat.com/show_bug.cgi?id=2274740

Proposed as a FE to get it fixed for F40 GA.

pcdubs, 2024-04-15:
New fedora-release included in today's Rawhide compose, confirmed service is now enabled:

`fdo-client.post: Created symlink /etc/systemd/system/multi-user.target.wants/fdo-client-linuxapp.service → /usr/lib/systemd/system/fdo-client-linuxapp.service.`

Full Logs https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-41-20240415.0/logs/aarch64/IoT/ostree-2/create-ostree-repo.log

pcdubs, 2024-05-22:
This was completed above, closing.

closed 2024-05-22 by pcdubs