fedora-iot/iot-distro #42: How to mount an ignition file into the simplified provision image?

odra, 2024-05-14:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
The docs  are not clear how to use a custom ignition file with the simplified provision installer.

Can I use a local file or just a remote one over http?

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Download the simplified provision iso
2. Run it using qemu-system
3. How to mount an ignition file ...?

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**

```
Using Fedora IoT 40, I can't login into it.
```

**Additional context**

As the "bug" description says, the docs is not very clear how to mount an ignition file into the image.

miabbott, 2024-05-17:
@mmartinv could you tackle this one?

odra, 2024-05-28:
Hello, any updates?

mmartinv, 2024-06-11:
@odra you need to edit the grub menu at boot time and append the following to the linux line:
```
coreos.inst.append=ignition.config.url=http://host/file.ign
```

The detailed steps are described in the [official documentation](https://docs.fedoraproject.org/en-US/iot/ignition-device-setup/#_edit_the_boot_parameters)