fedora-iot/iot-distro #44: need to provide download link to simpified-provisioner

miabbott, 2024-05-21:
The [docs](https://docs.fedoraproject.org/en-US/iot/ignition-device-setup/) for the `simplified-provisioner` point to iot.fedoraproject.org to download the artifact, but there is no link to the artifact there.

I suspect this requires a change outside of docs.fedoraproject.org, but maybe in the short-term we provide a dirty link to https://mirror.fcix.net/fedora-alt/iot/40/IoT/x86_64/iso/ (and `aarch64` version) for folks to discover the artifact.

nullr0ute, 2024-05-21:
We need to file a ticket with websites to get it added to https://fedoraproject.org/iot/download

miabbott, 2024-05-21:
> We need to file a ticket with websites to get it added to https://fedoraproject.org/iot/download

Done! - https://gitlab.com/fedora/websites-apps/fedora-websites/fedora-websites-3.0/-/issues/298

nullr0ute, 2024-11-13:
We have a link, naming could be improved.

closed 2024-11-13 by nullr0ute