fedora-iot/iot-distro #50: Install Image aarch64 Kernel Memory Leak

lethedata, 2024-06-10:
**Describe the bug**
The current install images hosted on [IoT Download Page](https://fedoraproject.org/iot/download) include the `6.8.5-301.fc40.aarch64` kernel which has a known memory leak leading to high memory usage and system crashes.

- Bugzilla Bug Report: [Bug 2275290 - memory leak in aarch64 beginning with kernel 6.8.4](https://bugzilla.redhat.com/show_bug.cgi?id=2275290)
- Forums Discussion: [High memory usage in F40 on RPi 4, unable to find which process used them](https://discussion.fedoraproject.org/t/high-memory-usage-in-f40-on-rpi-4-unable-to-find-which-process-used-them/114598)

**To Reproduce**
1. Install Fedora IoT release image on an aarch64 system
    - Note: Systems with low memory such as 1GB rpi3 have system crashes in a few hours
2. Do not update
3. Monitor for increasingly high memory usage

**Expected behavior**
Normal non-leaking memory usage preventing system crashes.

**Additional context**
Updating fixes the issue as the updates include an updated kernel. Currently `6.8.11-300.fc40.aarch64`.

nullr0ute, 2024-06-10:
>     2. Do not update

This is the key point, the bug has long been fixed by an update, as have *numerous* CVEs, so while it's a bug, it is fixed.

closed 2024-06-10 by nullr0ute

miabbott, 2024-06-10:
@nullr0ute do we ever entertain rebuilding the install media after GA in the case of a known issue?

nullr0ute, 2024-06-10:
> @nullr0ute do we ever entertain rebuilding the install media after GA in the case of a known issue?

We rebuild them daily, we do on occasion, it use to be monthly but that's slipped of late, republish them but that requires a full QE because there's *always* known issues, some like this one are easily fixed with an upgrade.

lethedata, 2024-06-10:
That makes sense to me.

The only argument I'd have for the need to republish *now* is that this known issue could be system crippling and prevent an upgrade. Tho the system I ran into this on was pretty constrained resource wise and was still able to update without issue so it 's probably fine to wait till the next publish.

nullr0ute, 2024-06-10:
Even on a RPi3 the memory leak wasn't bad enough you couldn't upgrade before you ran out of memory.

But we do have a process to do updates of releases on the website, we were planning on doing one RSN.