fedora-iot/iot-distro #52: Enabling composefs by default in IoT (Fedora 41 Change)

pcdubs, 2024-06-19:
From the F41 change:

Enabling `composefs` by default makes the root mount of the system (/) a truly read only filesystem, increasing the system integrity and robustness. This is the first step toward a full *at runtime* verification of filesystem integrity.

Full details:
https://fedoraproject.org/wiki/Changes/ComposefsAtomicCoreOSIoT

See:
https://github.com/containers/composefs
https://docs.kernel.org/filesystems/erofs.html
https://www.kernel.org/doc/Documentation/filesystems/overlayfs.txt
https://blogs.gnome.org/alexl/2024/01/15/testing-composefs-in-silverblue/

travier, 2024-07-17:
This has been approved by FESCo: https://pagure.io/fesco/issue/3240

miabbott, 2025-03-26:
Some of this landed in Rawhide and F42

https://pagure.io/fedora-iot/ostree/pull-request/74
https://pagure.io/fedora-iot/ostree/pull-request/75

There's a possibility of an issue migrating to static GRUB configs (https://github.com/fedora-iot/iot-distro/issues/83), but we have not seen a real world report of this from IoT users.

closed 2025-06-16 by pcdubs