fedora-iot/iot-distro #53: Fedora IoT: Roadmap to Bootable containers

pcdubs, 2024-06-19:
## Notes
* This is a work in progress and in many cases mirrors the other ostree based Fedora editions- CoreOS and AtomicDesktops.

## Roadmap
### Building and publishing Bootable Container images
* We are currently building a `tier-0` container and a `tier-1` containers as a part of the Fedora IoT compose process.
* `tier-1` images are published to:
  * [Quay.io](https://quay.io/repository/fedora/fedora-bootc?tab=tags)
  * registry.fedoraproject.org/fedora-bootc

### bootupd integration
* [Fedora 41 change](https://fedoraproject.org/wiki/Changes/FedoraSilverblueBootupd)
* [x] include `bootupd` in the IoT compose
  * [f41: enable bootupd](https://pagure.io/fedora-iot/ostree/c/4ef41ffa05c6973590756ea4307d8899904e46b9?branch=main)
* Testing the latest in Fedora 41, ensuring issues reported in [Fedora 40](https://github.com/rhinstaller/anaconda/pull/5508) are resolved.

### bootc integration
* [Fedora 41 change](https://fedoraproject.org/wiki/Changes/DNFAndBootcInImageModeFedora)
* [x]  include `bootc` in the IoT compose
  * [F41: add bootc to packages](https://pagure.io/fedora-iot/ostree/c/c1ecf382b89fcdcc85ddbca1fd2dfc1b3123c762?branch=main)
* include `rpm-ostree` until all issues resolved
* Bootc is currently root only: no unprivileged interface, no DBus interface
* Related discussions:
  * [cli: Add --json support to --check containers/bootc#472](https://github.com/containers/bootc/pull/472)
  * [`bootc status` does not report unlocked status containers/bootc#474](https://github.com/containers/bootc/issues/474)

### DNF5 integration
* [Fedora 41 change](https://fedoraproject.org/wiki/Changes/DNFAndBootcInImageModeFedora)
* [ ]  Include `dnf5` in the IoT compose:
  * https://github.com/fedora-iot/iot-distro/issues/51

### Switching to Bootable Container images
* [Fedora IoT derived bootc image is defined and composed automatically](https://github.com/fedora-iot/iot-distro/issues/19)
* Migrate Fedora IoT users to update via a container image by default: issue to be created

#### Local package layering
* Discussion of how users can locally layer packages
  * Tracked in https://gitlab.com/fedora/bootc/tracker/-/issues/4

### Rebasing on Fedora Bootc manifests
* To be investigated

### Rebasing on Fedora Bootc container images
* Better Fedora Infrastructure support for container builds (not yet tracked)
* Support for container deltas / zstd:chunked:
  * [Add support for zstd:chunked ostreedev/ostree-rs-ext#608](https://github.com/ostreedev/ostree-rs-ext/issues/608)
  * https://gitlab.com/fedora/bootc/tracker/-/issues/9
* Investigate the new Experimental Base Images Builder: https://gitlab.com/fedora/bootc/base-images-experimental

### Osbuild
* Investigate changes needed to `osbuild-composer` for IoT deliverables
  * iot-raw-image
  * iot-installer
  * iot-commit
  * iot-simplified-provisioning

## Documentation updates
* Updates to https://docs.fedoraproject.org/en-US/iot/getting-started/

## References
 See:

 * [Roadmap to Fedora Bootable Containers](https://gitlab.com/fedora/ostree/sig/-/issues/26)
 * [Fedora CoreOS bootc roadmap](https://github.com/coreos/fedora-coreos-tracker/issues/1726)
 * https://fedoraproject.org/wiki/Initiatives/Fedora_bootc

travier, 2024-06-20:
> * registry.fedoraproject.org/fedora-bootc

Those are the bootc ones. Don't you want to have IoT dedicated ones?

pcdubs, 2024-06-20:
> > * registry.fedoraproject.org/fedora-bootc
> 
> Those are the bootc ones. Don't you want to have IoT dedicated ones?

Yes, covered a little further down:

> * Fedora IoT derived bootc image is defined and composed automatically

adelton, 2024-07-01:
Does this issue effectively replace https://github.com/fedora-iot/iot-distro/issues/38?

miabbott, 2024-09-05:
For broader awareness, I filed an issue in the Fedora bootc project about the size of the base image - https://gitlab.com/fedora/bootc/tracker/-/issues/36

nullr0ute, 2024-11-27:
I believe this is now complete

closed 2024-11-27 by nullr0ute

travier, 2024-11-27:
Well, we now have both but I don't think we've completed the migration.

miabbott, 2025-01-08:
We haven't migrated to bootable containers, so I am going to reopen this

nullr0ute, 2025-01-09:
I read "Roadmap to Bootable containers" as getting them supported, not having them as the only option :)