fedora-iot/iot-distro #54: Boot fails with "vmlinuz has invalid signature" or "bad shim signature, you need to load the kernel first"

miabbott, 2024-06-20:
This is just a copy of a similar issue that is affecting other ostree-based systems, specifically the Atomic Desktops. See https://github.com/fedora-silverblue/issue-tracker/issues/543 for full details.

The idea is that once the adoption of `bootupd` happens (#6), we will be able to avoid these kinds of problems in the future.

---

## Suggested Workaround

(from https://github.com/fedora-silverblue/issue-tracker/issues/543#issuecomment-2048350047)

**Warning: These instructions should be safe to follow, but still, do at your own risk, make backups**

```
# Enter a root shell on the host (i.e. not in a toolbox)
$ sudo -i

# Make a backup of the content of the EFI partition
$ cd /boot/efi/
$ cp -a EFI EFI.bkp

# Copy updated bootloader versions
$ cp /usr/lib/ostree-boot/efi/EFI/BOOT/{BOOTIA32.EFI,BOOTX64.EFI,fbia32.efi,fbx64.efi} /boot/efi/EFI/BOOT/
$ cp /usr/lib/ostree-boot/efi/EFI/fedora/{BOOTIA32.CSV,BOOTX64.CSV,grubia32.efi,grubx64.efi,mmia32.efi,mmx64.efi,shim.efi,shimia32.efi,shimx64.efi} /boot/efi/EFI/fedora/

# Only needed if it exists already on your system
$ cp /usr/lib/ostree-boot/efi/EFI/fedora/shimx64.efi /boot/efi/EFI/fedora/shimx64-fedora.efi

# Sync changes to the disk
$ sync

# Reboot
```

Once reboot is successful, you can remove the backup copies:

```
# Enter a root shell on the host (i.e. not in a toolbox)
$ sudo -i

# Make a backup of the content of the EFI partition
$ cd /boot/efi/
$ rm -ri ./EFI.bkp

# Sync changes to the disk
$ sync
```

Edit: Updated to add 32bits EFI binaries as well.

For aarch64, update the filenames as needed.

miabbott, 2024-07-01:
The introduction of the 6.9 kernel in Fedora seems to be the trigger for this issue (along with having an older `shim`/bootloader).  This happened as part of 40.20240617.0 (`62c8ff246886838c8b5df7ca5ff060fccee8705fa7114f3ec47dad0103ac3ba9`) on the `fedora/stable/x86_64/iot` ref.

Affected users should follow the workaround instructions noted above.

nullr0ute, 2024-11-27:
We now have a new shim with appropriate signed bits and all the various work arounds for upgrade paths.

closed 2024-11-27 by nullr0ute