fedora-iot/iot-distro #59: F40: missing `os` directory on dl.fp.o for the release

miabbott, 2024-09-11:
The F39 download directory for IoT on dl.fedoraproject.org includes three subdirs:  `images`, `iso`, `os`

https://dl.fedoraproject.org/pub/alt/iot/39/IoT/aarch64/

The F40 download directory for IoT on dl.fp.o only has two subdirs: `images`, `iso`

https://dl.fedoraproject.org/pub/alt/iot/40/IoT/aarch64/

What happened to the `os` subdir?  Is there rationale for it's omission?

pcdubs, 2024-09-12:
This was a result of moving to osbuild for our [deliverables](https://github.com/fedora-iot/iot-distro/issues/24) . We decided to document how to use [pxe](https://github.com/fedora-iot/iot-distro/issues/28) .

miabbott, 2024-09-12:
Ah right. So this is basically a duplicate of #24.

closed 2024-09-12 by miabbott