fedora-iot/iot-distro #60: remove `dnsmasq` from the default package set

miabbott, 2024-09-20:
We saw a [report in the RHEL for Edge downstream](https://issues.redhat.com/browse/THEEDGE-4012) about how the inclusion of `dnsmasq` in the OS was causing issues for CIS Level 1 server compliance.  (Example audit rule for CIS compliance - https://www.tenable.com/audits/items/CIS_Oracle_Linux_9_v2.0.0_L1_Server.audit:35758e8a3ea48aff90e79b2d49199741)

The git archeology for the inclusion of that package appears to go back to the first definition of the IoT package set - https://pagure.io/fedora-iot/ostree/c/e4dff0e6fb85568963701ced7e8ca97e88e5f74a?branch=master

This looks very similar to the base content set that was used for Fedora Atomic Host and the rationale for the inclusion of `dnsmasq` in Atomic Host had to do with supporting `openshift-ansible`/OpenShift 3.11 - https://www.pagure.io/fedora-atomic/c/0435250a9308ba55e56f3c31edfe63aab2c41d8d

An inspection of the Requires/Recommends on `dnsmasq` doesn't show any default packages that depend on the package:

```
[core@localhost ~]$ rpm -q --whatrequires dnsmasq
no package requires dnsmasq
[core@localhost ~]$ rpm -q --whatrecommends dnsmasq
no package recommends dnsmasq
```

Given the history and lack of dependencies on the package, it seems like a good candidate for removal from IoT.

(We already decided to remove it from future versions of RHEL for Edge - https://github.com/osbuild/images/pull/921)

nullr0ute, 2024-09-20:
Yes, makes sense, from (a very vague) memory it was originally needed for container networking, but that's no longer the case. Filed this PR: https://pagure.io/fedora-iot/ostree/pull-request/62

pcdubs, 2024-09-20:
Merged.

closed 2024-09-20 by pcdubs