fedora-iot/iot-distro #63: F42 IoT installation fails as of Fedora-IoT-42-20240927.1

pcdubs, 2024-10-01:
**Describe the bug**
F42 IoT installation fails as of Fedora-IoT-42-20240927.1

See - https://openqa.fedoraproject.org/tests/2918504#step/_console_wait_login/13

**To Reproduce**
Attempt installation of Fedora 42, on boot:

```
Detected no PKCS#11 device, retry PKCS#11 detection? [yY/nN] 
[  339.708878] dracut-initqueue[2893]: Failed to query password: Timer expired
[  339.746926] dracut-initqueue[3248]: loading "libpcsclite_real.so.1" failed: libpcsclite_real.so.1: cannot open shared object file: No such file or directory
```

miabbott, 2024-10-01:
Quick search of that error only comes up with - https://github.com/NixOS/nixpkgs/issues/309158

Not quite the same, but might narrow things down.

Do we have a package diff between last known good and this failing version?

pcdubs, 2024-10-01:
Upgrade from last working compose (42-20240925.0) 

```
Upgraded:
  bootupd 0.2.21-1.fc42 -> 0.2.23-1.fc42
  ca-certificates 2024.2.69_v8.0.401-1.fc42 -> 2024.2.69_v8.0.401-2.fc42
  clevis 20-3.fc41 -> 21-1.fc42
  clevis-dracut 20-3.fc41 -> 21-1.fc42
  clevis-luks 20-3.fc41 -> 21-1.fc42
  clevis-systemd 20-3.fc41 -> 21-1.fc42
  coreutils 9.5-9.fc42 -> 9.5-10.fc42
  coreutils-common 9.5-9.fc42 -> 9.5-10.fc42
  criu 3.19-7.fc41 -> 4.0-1.fc42
  criu-libs 3.19-7.fc41 -> 4.0-1.fc42
  curl 8.10.1-1.fc42 -> 8.10.1-2.fc42
  fwupd 1.9.23-1.fc41 -> 1.9.25-1.fc42
  fwupd-plugin-modem-manager 1.9.23-1.fc41 -> 1.9.25-1.fc42
  fwupd-plugin-uefi-capsule-data 1.9.23-1.fc41 -> 1.9.25-1.fc42
  glib2 2.82.1-1.fc42 -> 2.82.1-2.fc42
  kernel 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  kernel-core 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  kernel-modules 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  kernel-modules-core 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  kernel-tools 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  kernel-tools-libs 6.12.0-0.rc0.20240924gitabf2050f51fd.11.fc42 -> 6.12.0-0.rc0.20240927git075dbe9f6e3c.14.fc42
  libcurl 8.10.1-1.fc42 -> 8.10.1-2.fc42
  libgcc 14.2.1-3.fc42 -> 14.2.1-4.fc42
  libgomp 14.2.1-3.fc42 -> 14.2.1-4.fc42
  libstdc++ 14.2.1-3.fc42 -> 14.2.1-4.fc42
  podman 5:5.2.2-2.fc42 -> 5:5.2.3-2.fc42
  selinux-policy 41.18-1.fc42 -> 41.19-1.fc42
  selinux-policy-targeted 41.18-1.fc42 -> 41.19-1.fc42
  tmux 3.4-2.fc41 -> 3.5-1.fc42
  tpm2-pkcs11 1.9.0-7.fc41 -> 1.9.1-1.fc42
Added:
  openpace-1.1.3-2.fc41.x86_64
  opensc-0.25.1-4.fc41.x86_64
  pcsc-lite-2.3.0-1.fc41.x86_64
  pcsc-lite-ccid-1.6.1-2.fc41.x86_64
  pcsc-lite-libs-2.3.0-1.fc41.x86_64
Run "systemctl reboot" to start a reboot
```

```
rpm -ql pcsc-lite-libs
...
/usr/lib64/libpcsclite.so.1
/usr/lib64/libpcsclite_real.so.1
```

pcdubs, 2024-10-01:
The `Clevis` upgrade added a dep on `pcsc-lite` and `opensc`

miabbott, 2024-10-14:
Fedora-IoT-42-20241011.0 looks fixed

https://openqa.fedoraproject.org/tests/2946497

closed 2024-10-14 by miabbott