fedora-iot/iot-distro #64: Manufacturing client does not work in Fedora IoT Simplified Provisioner 40.

mmartinv, 2024-10-02:
**Describe the bug**

**To Reproduce**
1. Boot the simplified provisioner and enter the corresponding FDO manufacturing cmdline arguments also add `coreos.inst.skip_reboot` to skip the poweroff once the installation has finished.
2. Once the image has been installed enter the emergency shell and check the status of the manufacturing client

**Expected behavior**
The manufacturing client started the DIUN process.

**Screenshots**
[image]

**OS version:**
Fedora 40

```
Please replace this line with output of `rpm-ostree status -b`
```

**Additional context**

It looks like the systemd generator is unable to write to `/etc`
[image]

closed 2024-10-07 by mmartinv

nullr0ute, 2024-10-07:
Has that fix landed?

miabbott, 2024-10-07:
@nullr0ute I believe we'll still need to do another release and package it for Fedora/CentOS/RHEL

nullr0ute, 2024-10-08:
For ref the upstream merge for FDO was actually https://github.com/fdo-rs/fido-device-onboard-rs/pull/677