fedora-iot/iot-distro #65: ignition provisioning using tftp fails

mlout, 2024-10-15:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
Ignition fails to provision when using a tftp server to serve the ignition file. My assumption is that ignition does not wait until NetworkManager is done setting up the connection.
Tested using both the Fedora-IoT-raw-40-20240422.3.x86_64.raw.xz (in a Virtualbox VM) and Fedora-IoT-raw-40-20240422.3.aarch64.raw.xz (flashed to a Compute Module 4).

**To Reproduce**
_I will describe the steps for x86_64, as that might be easier to reproduce._
Please describe the steps needed to reproduce the bug:
1. Setup a TFTP server on your host machine, I used this docker image: https://github.com/taskinen/tftp and verified that the TFTP server was working by using a TFTP client on another machine.
3. Setup a Virtualbox VM with 'Bridged Adapter' networking, so that the VM is able to reach the TFTP server.
4. Convert the image for use with VirtualBox
`
xz --decompress Fedora-IoT-raw-40-20240422.3.x86_64.raw.xz
VBoxManage convertfromraw Fedora-IoT-raw-40-20240422.3.x86_64.raw Fedora-IoT.vdi
`
5. Attach the vdi file as disk to the VM
6. Start the VM, press 'e' at the grub bootloader, add the following to the linux line: ip=dhcp rd.neednet=1 ignition.config.url=tftp://192.168.1.150/fiot.ign , press C-x to boot

**Expected behavior**
Ignition downloads fiot.ign from the tftp server and executes it's contents.
Instead it fails immediately with:
`
[    5.261072] localhost.localdomain ignition[476]: failed to fetch config: write udp [::]:52430->192.168.1.150:69: sendto: network is unreachable
[    5.261658] localhost.localdomain ignition[476]: failed to acquire config: write udp [::]:52430->192.168.1.150:69: sendto: network is unreachable
[    5.261696] localhost.localdomain ignition[476]: Ignition failed: write udp [::]:52430->192.168.1.150:69: sendto: network is unreachable
`

**Screenshots**
See attached [rdsosreport.txt](https://github.com/user-attachments/files/17374477/rdsosreport.txt) instead.

**OS version:**
Fedora-IoT-raw-40-20240422.3.x86_64.raw.xz
Fedora-IoT-raw-40-20240422.3.aarch64.raw.xz

**Additional context**
When changing the TFTP server to a HTTP server (using python3 -m http.server) and changing the ignition.config.url to use http, provisioning works. I can see ignition-fetch.service and nm-wait-online-initrd.service take more time during booting.

nullr0ute, 2024-11-07:
Does it work if you serve it from a HTTP connection?

nullr0ute, 2024-11-07:
I'm not sure how ignition supports tftp but I suspect we would need to ship a tftp client and the world tends to be moving away from tftp because it's generally regarded as insecure.

miabbott, 2024-11-07:
> I'm not sure how ignition supports tftp but I suspect we would need to ship a tftp client and the world tends to be moving away from tftp because it's generally regarded as insecure.

https://github.com/coreos/ignition/blob/a204f429f13194ae379be9401d49e5241439660b/internal/resource/url.go#L231

Looks like it ultimately uses https://github.com/pin/tftp

Hmm...it looks like HTTP supports retries, where as TFTP does not - https://github.com/coreos/ignition/issues/711

That may explain the difference in success here.

nullr0ute, 2024-11-07:
So a lot of firewalls block tftp, the error above has `sendto: network is unreachable
` so that says to be the service isn't contactable, tftp is quite problematic because it's not authenticated and can't run over TLS or be put through proxies so it's very easy to MITM it. The network (or possibly service) is unreachable so that says to me a firewall/routing or some other local problem and not something that we can actually resolve.

mlout, 2024-11-08:
> Does it work if you serve it from a HTTP connection?

Yes, HTTP works. I started first with TFTP because I assumed setting up a HTTP server would be a lot or work, but in reality it isn't. So for me it is fine to switch to HTTP.

> So a lot of firewalls block tftp, the error above has `sendto: network is unreachable ` so that says to be the service isn't contactable, tftp is quite problematic because it's not authenticated and [...]

From another host I was able to contact the TFTP server (with a TFTP client), so that is why I am suspecting something in the Fedora image. Also the lack of delay (for what I think is the DHCP configuration) compared to when using HTTP is quite noticeable. But as stated earlier, switching to HTTP is fine for me.

nullr0ute, 2024-11-13:
> From another host I was able to contact the TFTP server (with a TFTP client), so that is why I am suspecting something in the Fedora image. Also the lack of delay (for what I think is the DHCP [...]

So you'd have DHCP for both HTTP or TFTP use case so I'm not sure it's that. I'm also not sure from what you say where the delay is, whether it's for TFTP or HTTP, and whether that delay is good or bad in the context.

Either way TFTP is tending to be retired due to numerous reasons so HTTP is definitely the way to go here.

closed 2024-11-13 by nullr0ute