fedora-iot/iot-distro #66: IoT 42 installer images since Fedora-IoT-42-20241018.0 boot to text mode with error "Wayland startup failed, falling back to text mode"

pcdubs, 2024-10-22:
**Describe the bug** 
From: https://pagure.io/fedora-iot/issue/61

anaconda was recently ported to Wayland. Apparently IoT needs some sort of adjustment for this (some package missing?), because since it happened, IoT images boot to text mode, with an error "Wayland startup failed, falling back to text mode". See e.g. https://openqa.fedoraproject.org/tests/2971030#

mmartinv, 2024-11-08:
After reproducing the problem it looks like `/usr/libexec/anaconda/run-in-new-session` is failing because `python3-pam` package is missing:
~~~
ov 08 10:10:40 localhost.localdomain anaconda[2222]: program: Running... /usr/libexec/anaconda/run-in-new-session --user root --service anaconda --session-type wayland --session-class user --vt 6 gnome-kiosk --sm-disable --wayland --no-x11 --wayland-display wl-sysinstall-0
Nov 08 10:10:40 localhost.localdomain org.fedoraproject.Anaconda.Modules.Timezone[2323]: DEBUG:anaconda.modules.timezone.timezone:Time sources are set to: []
Nov 08 10:10:40 localhost.localdomain anaconda[2222]: anaconda: core.threads: Thread Done: AnaWaitForConnectingNMThread (140089852163776)
Nov 08 10:10:40 localhost.localdomain gnome-kiosk[2433]: Traceback (most recent call last):
Nov 08 10:10:40 localhost.localdomain gnome-kiosk[2433]:   File "/usr/libexec/anaconda/run-in-new-session", line 24, in <module>
Nov 08 10:10:40 localhost.localdomain gnome-kiosk[2433]:     import pam
Nov 08 10:10:40 localhost.localdomain gnome-kiosk[2433]: ModuleNotFoundError: No module named 'pam'
~~~

miabbott, 2024-11-08:
@mmartinv I'm a bit confused by the fix...

I thought the Anaconda ISO for IoT 42 is built by osbuild, not b-i-b?

miabbott, 2024-11-08:
Also, if `anaconda-core` requires `python3-pam`, shouldn't we also be fixing this in the spec file for anaconda?

mmartinv, 2024-11-08:
> @mmartinv I'm a bit confused by the fix...
> 
> I thought the Anaconda ISO for IoT 42 is built by osbuild, not b-i-b?

My fault, I was able to reproduce the same issue while creating the anaconda ISO with b-i-b and I assumed the image was also built with b-i-b which is obviously not the case.

miabbott, 2024-11-08:
> Well, I don't think the problem is actually fixed in rawhide: The `python3-pam` dependency was added to the `anaconda-install-img-deps` package and in my opinion it should have been added to `anaconda-core` package because it's the package that contains `/usr/libexec/anaconda/run-in-new-session`.
> 
> I think we need to discuss this with anaconda guys.

Good call out; I missed that the `Requires` was for a different sub-package.

Looks like the [Anaconda folks use Bugzilla](https://github.com/rhinstaller/anaconda?tab=readme-ov-file#i-want-to-report-a-bug), so let's open an bug with them to discuss.

mmartinv, 2024-11-11:
@pcdubs I am a bit confused, how is Fedora-IoT-42 anaconda ISO generated? I don't see fedora-42 nor fedora-rawhide in the list of osbuild supported images

achilleas-k, 2024-11-11:
> I don't see fedora-42 nor fedora-rawhide in the list of osbuild supported images

If you define a repository configuration in osbuild-composer as `fedora-42.json`, it will enable building F42 images using the Fedora definitions (with the version range checks in the code).  If there's any special conditions needed for F42 (package list changes, for example), they need to be added in the `pkg/distro/fedora/` package in osbuild/images.

closed 2024-12-03 by mmartinv