fedora-iot/iot-distro #69: Document how to use the Fedora IoT bootable container images

pcdubs, 2024-10-23:
Document how to use the Fedora IoT bootable container images and it to our [getting started guide](https://docs.fedoraproject.org/en-US/iot/getting-started/)

* Document how to use [bootc-image-builder](https://github.com/osbuild/bootc-image-builder) to create anaconda installers, disk images and qcow2 images. 
* Document how to upgrade to image mode (bootable containers) for upgrades for existing users. 
* When needed, link to the existing upstream documentation for additional information and use cases.

lucamaf, 2025-01-16:
what's the status on this one? It would be nice to finally have an easy and fast way to build locally customized fedora iot images

gucci-on-fleek, 2025-01-16:
@lucamaf

> what's the status on this one? It would be nice to finally have an easy and fast way to build locally customized fedora iot images

Not documentation, but here's what I'm using at the moment:

https://github.com/gucci-on-fleek/maxchernoff.ca/tree/802a13e/builder/containers/fedora-iot
https://github.com/gucci-on-fleek/maxchernoff.ca/blob/802a13e/builder/.config/systemd/user/container%40fedora-iot.service

I've got it set to build daily, and it usually takes less than 5 minutes to run.

lucamaf, 2025-01-23:
how would I go about selecting arch64 of fedora-iot during build process?

say-paul, 2025-05-26:
@lucamaf  use tags to select the arch: https://quay.io/repository/fedora/fedora?tab=tags