fedora-iot/iot-distro #70: User SSH authorized keys are not added by ignition

mlout, 2024-10-30:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
When using the following ignition config (generated from a Butane file), with the image Fedora-IoT-raw-41-20241027.0.aarch64.raw.xz on a Raspberry Pi CM4:
```
{
  "ignition": {
    "version": "3.4.0"
  },
  "passwd": {
    "users": [
      {
        "groups": [
          "wheel"
        ],
        "name": "firmware",
        "passwordHash": "redacted",
        "sshAuthorizedKeys": [
          "ssh-ed25519 redacted"
        ]
      }
    ]
  },
  "storage": {
    "files": [
      {
        "path": "/etc/hostname",
        "contents": {
          "compression": "",
          "source": "data:,redacted"
        },
        "mode": 420
      }
    ]
  }
}
```
The hostname is set, the firmware user is created with the desired password, but the SSH key does not seem to be added anywhere. When trying to login I see my SSH client offering the public key, but then I still get asked for the password. So I think the public key was never deployed to the device.

When logging in to the device (with password) I get the following message:
`Could not chdir to home directory /var/home/firmware: No such file or directory`
Maybe this is related, since the authorized_keys file is usually placed inside the users home directory.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Serve the ignition file above as fiot.ign using a webserver on a different device
2. Modify the Fedora-IoT-raw-41-20241027.0.aarch64.raw.xz image, adding "ip=dhcp rd.neednet=1 ignition.config.url=http://192.168.37.54/fiot.ign" to ignition.firstboot
3. Flash the device using arm-image-installer on a Fedora workstation host with the following command: sudo ./arm-image-installer --image=../Fedora-IoT-raw-41-20241027.0.aarch64.raw.xz --media=/dev/sda --resizefs --target=rpi4 --selinux off
4. Boot the device

**Expected behavior**
When provisioning is finished, I am able to login as the firmware user with the SSH key

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**
Fedora-IoT-raw-41-20241027.0.aarch64.raw.xz

**Additional context**
redacted SSH client output:
```
debug1: Offering public key: redacted agent
debug1: Authentications that can continue: publickey,gssapi-keyex,gssapi-with-mic,password
debug1: Trying private key: /home/user/.ssh/id_rsa
debug1: Trying private key: /home/user/.ssh/id_ecdsa
debug1: Trying private key: /home/user/.ssh/id_ecdsa_sk
debug1: Trying private key: /home/user/.ssh/id_ed25519
debug1: Trying private key: /home/user/.ssh/id_ed25519_sk
debug1: Trying private key: /home/user/.ssh/id_xmss
debug1: Trying private key: /home/user/.ssh/id_dsa
debug1: Next authentication method: password
firmware@192.168.52.82's password:
```

mlout, 2024-10-31:
I have the same results when using Fedora-IoT-raw-41-20241027.0.x86_64.raw.xz image inside a VM.

fdesjardins, 2024-10-31:
I ran into this issue as well with a non-root user. It worked fine for root.

I haven't tested this yet, but I think you may need to mount /home in the ignition file for non-root users. See https://opensuse.github.io/fuel-ignition/edit in the Add user and Mount /home sections. Granted that's for SUSE but I suspect it's a similar issue.

miabbott, 2024-10-31:
I was able to reproduce this with the Fedora 40 (Fedora-IoT-raw-40-20240422.3.x86_64.raw) raw disk image, as well.

And to be honest, I don't think we have ever tested using Ignition with the raw disk, nor have we ever supported that combination.

We have only supporte/tested Ignition with the use of the simplified-provisioner - https://docs.fedoraproject.org/en-US/iot/ignition-device-setup/

@runcom can you weigh in here?

nullr0ute, 2024-10-31:
With the raw image we don't have a separate /home directory, is that something that ignition "expects"?

miabbott, 2024-11-01:
> With the raw image we don't have a separate /home directory, is that something that ignition "expects"?

Not sure. It eventually shells out to `useradd` and `usermod`:

```
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(1): [started]  creating or modifying user "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(1): executing: "useradd" "--root" "/sysroot" "--create-home" "--password" "$y$j9T$HsJiToYGxDpKaKlkV1ai9.$ypj54d7KIKYSUMAFRBQ9SRoacZ6ZHc5f0Baobybz.iA" "--groups" "wheel" "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(1): [finished] creating or modifying user "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(2): [started]  setting password for "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(2): executing: "usermod" "--root" "/sysroot" "--password" "$y$j9T$HsJiToYGxDpKaKlkV1ai9.$ypj54d7KIKYSUMAFRBQ9SRoacZ6ZHc5f0Baobybz.iA" "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(2): [finished] setting password for "core"
Nov 01 12:08:03 localhost.localdomain ignition[709]: files: ensureUsers: op(3): [started]  adding ssh keys to user "core"
Nov 01 12:08:03 localhost.localdomain (ignition)[709]: wrote ssh authorized keys file for user: core
```

I would have expected `useradd` to create the necessary directories, but I may be mistaken.

Adding to this, `zezere-ignition` also runs after switching into the real root and tries to create the user, but fails:

```
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : Ignition 2.19.0
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : Stage: fetch
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : no config dir at "/usr/lib/ignition/base.d"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : no config dir at "/usr/lib/ignition/base.platform.d/file"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: DEBUG    : parsed url from cmdline: "http://192.168.122.147/ignition.json"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : GET http://192.168.122.147/ignition.json: attempt #1
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : GET result: OK
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: DEBUG    : parsing config with SHA512: f8bcfd621e49b3d11a61df6b4858cc83ec590c3dc6df7345711132d25c353b17cdc9c46371ec3500160796e5a62b6c9eac55bff07c7b301ee52d6c5216441e84
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : fetch: fetch complete
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : fetch: fetch passed
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched user config from "cmdline"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched base config from "system"
Nov 01 12:08:18 localhost.localdomain ignition[1456]: fetched user config from "cmdline"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1456]: INFO     : Ignition finished successfully
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : Ignition 2.19.0
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : Stage: disks
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : no config dir at "/usr/lib/ignition/base.d"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : no config dir at "/usr/lib/ignition/base.platform.d/file"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : disks: disks passed
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1462]: INFO     : Ignition finished successfully
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : Ignition 2.19.0
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : Stage: mount
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : no config dir at "/usr/lib/ignition/base.d"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : no config dir at "/usr/lib/ignition/base.platform.d/file"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : mount: mount passed
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1468]: INFO     : Ignition finished successfully
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1473]: INFO     : Ignition 2.19.0
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1473]: INFO     : Stage: files
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1473]: INFO     : no config dir at "/usr/lib/ignition/base.d"
Nov 01 12:08:18 localhost.localdomain zezere-ignition[1473]: INFO     : no config dir at "/usr/lib/ignition/base.platform.d/file"
[... cut ...]
drwxr-xr-x.  2 root root system_u:object_r:var_t:s0            4096 Oct 27 21:58 srv
drwxrwxrwt.  9 root root system_u:object_r:tmp_t:s0            4096 Nov  1 12:09 tmp
drwxr-xr-x. 11 root root system_u:object_r:usr_t:s0            4096 Nov  1 12:08 usrlocal
drwxr-xr-x.  2 root root system_u:object_r:var_yp_t:s0         4096 Nov  1 12:08 yp
$ ls -laZ /var/home
total 8
drwxr-xr-x.  2 root root system_u:object_r:home_root_t:s0 4096 Oct 27 21:58 .
drwxr-xr-x. 23 root root system_u:object_r:var_t:s0       4096 Nov  1 12:08 ..
```

miabbott, 2024-11-01:
I'm seeing similar problems using the simplified-provisioner as well.

fdesjardins, 2024-11-01:
Confirming the same for `Fedora-IoT-raw-41-20241027.0.x86_64.raw`, ignition won't create the user's home directory:

```
Nov 01 19:28:57 localhost.localdomain ignition[688]: files: ensureUsers: op(4): executing: "useradd" "--root" "/sysroot" "--create-home" "--password" "$2a$10$kfEn0Uog0xFT0C2BTzd0YuCHSyA4KO9rkqICzkdBRRDjyxuExzTE6" "user1"
Nov 01 19:28:58 localhost.localdomain ignition[688]: files: ensureUsers: op(4): [finished] creating or modifying user "user1"
Nov 01 19:28:58 localhost.localdomain ignition[688]: files: ensureUsers: op(5): [started]  setting password for "user1"
Nov 01 19:28:58 localhost.localdomain ignition[688]: files: ensureUsers: op(5): executing: "usermod" "--root" "/sysroot" "--password" "$2a$10$kfEn0Uog0xFT0C2BTzd0YuCHSyA4KO9rkqICzkdBRRDjyxuExzTE6" "user1"
Nov 01 19:28:58 localhost.localdomain ignition[688]: files: ensureUsers: op(5): [finished] setting password for "user1"
Nov 01 19:28:58 localhost.localdomain ignition[688]: files: ensureUsers: op(6): [started]  adding ssh keys to user "user1"
Nov 01 19:28:58 localhost.localdomain (ignition)[688]: wrote ssh authorized keys file for user: user1
```

```
Nov 01 19:29:12 localhost.localdomain zezere-ignition[1461]: CRITICAL : Ignition failed: failed to create users/groups: failed to configure users: failed to add keys to user "user1": failed to set SSH key: creating parent dirs for "/var/home/user1/.ssh/authorized_keys": permission denied
```

Interestingly, I logged in as `root` and manually created the `/var/home/user1/.ssh` directory and set the owner, `zezere-ignition` actually did create the `authorized_keys` file.

I tried to add a `storage.directories` section in the ignition file to do the same but no luck.

djach7, 2024-11-15:
Started investigating this, I was able to reproduce it and I found/learned a couple things:
- ssh keys should be written to `/.ssh/authorized_keys.d/`
- Moving the ssh keys should be handled by the `ssh-key-dir` rpm during boot (https://github.com/coreos/ssh-key-dir)
- The `ignition-edge` rpm is needed to make it all work (https://github.com/fedora-iot/ignition-edge)
- Both the `ssh-key-dir` rpm and the `ignition-edge` rpm are missing from the Fedora IoT 41 raw image

I'll continue with trying to figure out where these rpms got lost along the way and if they're the only issue

miabbott, 2024-11-15:
https://github.com/osbuild/images/blob/0584c20b0c14a89e833b875252114ceff61805e8/pkg/distro/fedora/package_sets.go#L123

https://github.com/osbuild/images/blob/0584c20b0c14a89e833b875252114ceff61805e8/pkg/distro/fedora/package_sets.go#L157

Both packages are present in the `iot-commit` code of `osbuild` :thinking:

pcdubs, 2024-11-15:
Neither is included in the commit created in koji - https://pagure.io/fedora-iot/ostree/blob/main/f/fedora-iot-base.yaml

We should check for other discrepancies.

djach7, 2024-11-19:
Submitted a pr to add the packages necessary to fix this here: https://pagure.io/fedora-iot/ostree/pull-request/66

I was able to login with ssh successfully and both a home directory and a .ssh directory were created, so I believe this is sorted out. Going to wait to close until the pr is merged.

djach7, 2024-11-19:
Closing now that PR is merged.

miabbott, 2024-11-20:
Let's keep this open until we get a working compose with the fix.

miabbott, 2025-02-02:
@djach7 Can you confirm that this fix has landed in a Fedora IoT compose?

djach7, 2025-02-03:
Yes, I can confirm that ignition is fully functional in F41 and rawhide as of the 1/31 compose. I'm going to close this issue.

closed 2025-02-03 by djach7