fedora-iot/iot-distro #71: Unable to boot into single user mode

miabbott, 2024-10-31:
**Describe the bug**
When appending the `single` keyword to the the `linux` kernel line of the boot entry in grub, the system boots into rescue mode, however you get the message `Cannot open access to console, the root account is locked`

**To Reproduce**
1. Boot Fedora IoT
2. Interrupt boot process at grub menu
3. Append `single` to the `linux` line
4. Continue boot

**Expected behavior**
I expect to be dropped into an interactive rescue shell.

**Screenshots**
[image]

**OS version:**

This happens with Fedora IoT 40 (40.20240923.1) and 41 (41.20241027.0)

**Additional context**
See also https://github.com/coreos/fedora-coreos-tracker/issues/1792

- You can workaround this problem by appending `init=/bin/sh` to the `linux` line of the grub entry

miabbott, 2024-10-31:
If you have a password configured for the `root` user, this does not happen.

We can't set a default `root` password and we can't enforce the creation of a `root` password when the system is provisioned, so maybe we just need some clear docs around this.

nullr0ute, 2024-10-31:
> If you have a password configured for the `root` user, this does not happen.

What doesn't happen? Do you mean it works as expected for the single key word?

Also is this a ostree issue, or a Fedora/systemd issue in general? I suspect it's a systemd feature (or bug depending on which way you look at it).

nullr0ute, 2024-10-31:
For reference this is not new, it's a systemd feature and not specific to ostree or even Fedora, it's also been the case for years, they added the ability to override it with an environment variable in 2018[1].

I think this should be, and it may already be but a quick search didn't bring anything up, documented as part of standard Fedora docs as it's not IoT (or ostree for that matter) specific.

[1] https://github.com/systemd/systemd/pull/10397

nullr0ute, 2024-10-31:
I'm going to close this, it's working as expected, and in fact is a feature we want for edge devices because else it's a security attack vector if a random person can access the root filesystem without any restriction.

If a user that wants to be able to do this without a password they can set the appropriate env variable, details are in the referenced systemd PR. It should be documented in general Fedora docs but it's already in systemd docs (I believe).

closed 2024-10-31 by nullr0ute

miabbott, 2024-10-31:
The feedback is fair and makes sense; I was applying use cases for desktops to the IoT/Edge environment, which don't have the same requirements.

nullr0ute, 2024-10-31:
It's useful to have the conversation documented here in a ticket though for people to see why a decision was made so thanks!