fedora-iot/iot-distro #72: missing `pcscd` command during `dracut-initqueue` service

miabbott, 2024-11-01:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
When the system is booted into the initrd, the `dracut-initqueue` service reports errors about a missing `pcscd` command.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Boot Fedora IoT 41 system
2. Observe journal during initrd stage

**Expected behavior**
No errors are observed during the `dracut-initqueue` service

**Additional context**

```
Nov 01 12:07:40 localhost.localdomain systemd[1]: Starting dracut-initqueue.service - dracut initqueue hook...
Nov 01 12:07:40 localhost.localdomain dracut-initqueue[624]: /lib/dracut/hooks/initqueue/settled/60-clevis-pkcs11-hook.sh: line 21: pcscd: command not found
Nov 01 12:07:40 localhost.localdomain dracut-initqueue[627]: grep: /etc/crypttab: No such file or directory
Nov 01 12:07:40 localhost.localdomain systemd[1]: Finished dracut-initqueue.service - dracut initqueue hook.
```

I don't know how serious/fatal this error is.  It appears that `pcscd` is provided by `pcsc-lite`, so perhaps we should consider including that package in IoT?

pcdubs, 2024-11-01:
We had it [briefly](https://github.com/fedora-iot/iot-distro/issues/63) last month, Clevis added a dep on `pcsc-lite` and `opensc`  but then changed to [recommended](https://src.fedoraproject.org/rpms/clevis/c/e5f0f49c587822fb994720205ae155b4b4f9bafd?branch=f41) to avoid pulling in Python, which we already have in IoT. 

FWIW, I had no issues with Clevis in F41 on the Fitlet2.

nullr0ute, 2024-11-06:
pcscd is for SmartCard support, I believe in the clevis context it's more for end user devices like laptops with SCs that are used for lock/unlock, I'm not sure we have a requirement/request there, clevis in the IoT context is primarily around TPM2 support.

closed 2024-11-13 by nullr0ute