fedora-iot/iot-distro #74: Unable to install any packages when building container based on IoT image

alaviss, 2024-11-11:
**Describe the bug**
Installing packages in a `Containerfile` with `fedora-iot` base image always fail when building with `podman`.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Run `podman build .` with this `Containerfile`:

   ```dockerfile
   FROM quay.io/fedora/fedora-iot:41
   RUN dnf -y install tailscale && dnf clean all
   ````

**Expected behavior**
The container is built successfully

**Build log**

<details>

```
STEP 1/2: FROM quay.io/fedora/fedora-iot:41
STEP 2/2: RUN dnf -y install tailscale
Updating and loading repositories:
 Fedora 41 - x86_64 - Updates           100% |   3.8 MiB/s |   4.3 MiB |  00m01s
 Fedora 41 openh264 (From Cisco) - x86_ 100% |   9.4 KiB/s |   4.8 KiB |  00m01s
 Fedora 41 - x86_64                     100% |   5.5 MiB/s |  35.5 MiB |  00m06s
Repositories loaded.
Package    Arch   Version       Repository      Size
Installing:
 tailscale x86_64 1.70.0-1.fc41 fedora      35.0 MiB

Transaction Summary:
 Installing:         1 package

Total size of inbound packages is 12 MiB. Need to download 12 MiB.
After this operation, 35 MiB extra will be used (install 35 MiB, remove 0 B).
[1/1] tailscale-0:1.70.0-1.fc41.x86_64  100% |   8.7 MiB/s |  11.5 MiB |  00m01s
--------------------------------------------------------------------------------
[1/1] Total                             100% |   7.6 MiB/s |  11.5 MiB |  00m02s
Running transaction
Importing PGP key 0xE99D6AD1:
 UserID     : "Fedora (41) <fedora-41-primary@fedoraproject.org>"
 Fingerprint: 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
 From       : file:///etc/pki/rpm-gpg/RPM-GPG-KEY-fedora-41-x86_64
The key was successfully imported.
[1/3] Verify package files              100% |  33.0   B/s |   1.0   B |  00m00s
[2/3] Prepare transaction               100% |  25.0   B/s |   1.0   B |  00m00s
[3/3] Installing tailscale-0:1.70.0-1.f 100% |   1.0 GiB/s |  35.0 MiB |  00m00s
>>> Unpack error: tailscale-0:1.70.0-1.fc41.x86_64

Transaction failed: Rpm transaction failed.
Error: building at STEP "RUN dnf -y install tailscale": while running runtime: exit status 1

```

</details>

**Additional context**
From entering the container, I extracted the following error log:

```
2024-11-11T21:06:18+0000 [2] INFO RPM callback install start "tailscale-0:1.70.0-1.fc41.x86_64" total 36668204
2024-11-11T21:06:18+0000 [2] ERROR [rpm] ima: could not apply signature on '/usr/bin/tailscale;673271ca': Operation not permitted
2024-11-11T21:06:18+0000 [2] ERROR [rpm] Plugin ima: hook fsm_file_prepare failed
2024-11-11T21:06:18+0000 [2] INFO RPM callback install stop "tailscale-0:1.70.0-1.fc41.x86_64" amount 36668204 total 36668204
2024-11-11T21:06:18+0000 [2] ERROR [rpm] unpacking of archive failed on file /usr/bin/tailscale;673271ca: cpio: (error 0x2)
2024-11-11T21:06:18+0000 [2] ERROR RPM callback unpack error "tailscale-0:1.70.0-1.fc41.x86_64"
2024-11-11T21:06:18+0000 [2] INFO RPM callback close file
2024-11-11T21:06:18+0000 [2] ERROR [rpm] tailscale-1.70.0-1.fc41.x86_64: install failed
```

I found that the following modification allows the container to be built successfully:

```dockerfile
FROM quay.io/fedora/fedora-iot:41
RUN dnf -y remove rpm-plugin-ima && \
  dnf -y install tailscale && \
  dnf clean all
```

nullr0ute, 2024-11-11:
see the [rpm bug Fedora bugzilla](https://bugzilla.redhat.com/show_bug.cgi?id=2292869) and fedora-iot is meant for hardware level deployment, not container based, else please better explain your usecase.

alaviss, 2024-11-12:
> fedora-iot is meant for hardware level deployment, not container based, else please better explain your usecase.

I'm building containers to explore the use of `bootc` and whether to migrate my systems over. I believe that this is a perfectly valid use case given that `bootc` doesn't support layering like `rpm-ostree`.

pcdubs, 2024-11-13:
Thanks for the report and debugging. Reopening so we can investigate further.

pcdubs, 2025-07-02:
We disabled IMA by default due to other [issues]( https://bugzilla.redhat.com/show_bug.cgi?id=2346265) when building the ostree. This no longer appears to be an issue as a result.