fedora-iot/iot-distro #79: enable GRUB static migration and composefs in rawhide/f42

miabbott, 2025-02-17:
As we move closer to the adoption of image mode, we need to keep pace with the rest of the ostree/image mode ecosystem and prepare IoT accordingly.

Recently the Atomic Desktops have enabled GRUB static config migration and the use of composefs in Rawhide/F42.

See https://pagure.io/workstation-ostree-config/pull-request/591 and https://pagure.io/workstation-ostree-config/pull-request/624

We need similar changes in Fedora IoT; probably need them in https://pagure.io/fedora-iot/ostree/ and in osbuild/images

miabbott, 2025-02-17:
Note: the freeze for F42 is Tues Feb 18, so if we want these changes made for F42, we need to act quickly

pcdubs, 2025-03-03:
Fixed with 

https://pagure.io/fedora-iot/ostree/pull-request/75
https://pagure.io/fedora-iot/ostree/pull-request/74

closed 2025-03-03 by pcdubs