fedora-iot/iot-distro #83: Static GRUB config migration does not handle GRUB configs stored in ESP

miabbott, 2025-03-12:
This is a tracking issue along the same lines of:

- https://discussion.fedoraproject.org/t/sb42-updates-fail-on-system-with-512mb-boot-128mb-boot-efi/145041
- https://github.com/coreos/bootupd/issues/857

The Atomic Desktops have a workaround proposed here:

- https://pagure.io/workstation-ostree-config/pull-request/630

However, since IoT does not have versioned streams, it doesn't seem possible to introduce this fix to IoT users as part of the normal update process (i.e. going from F41 -> F42).

It may be possible for users to manually edit `/etc/default/grub` and add `GRUB_ENABLE_BLSCFG=true` to that file as a workaround, but it's not clear to me if that would be effective if the system is already at F42.

Otherwise, we would have to recommend IoT users reprovision their systems with F42 media.

miabbott, 2025-03-12:
@travier you have a better understanding of this issue than I do, so if you have corrections, please let us know.

miabbott, 2025-03-12:
Note: I've not hit this issue nor have we heard of any IoT users hitting this problem, so it is purely theoretical at this point.

travier, 2025-03-12:
> It may be possible for users to manually edit `/etc/default/grub` and add `GRUB_ENABLE_BLSCFG=true` to that file as a workaround, but it's not clear to me if that would be effective if the system is already at F42.

That should indeed happen before the update to F42.

We should probably try to have that features in the release notes or update blog post on the Fedora Magazine.

travier, 2025-06-27:
There is another option here that could fix that for IoT users: https://github.com/coreos/bootupd/issues/857#issuecomment-3013257311