fedora-iot/iot-distro #86: OpenQA versioning tests failing

djach7, 2025-03-28:
**Describe the bug**
OpenQA versioning tests are failing with a previous version of Fedora IoT 41 and 43. The tests are expecting the current date but receiving yesterday's date.

**To Reproduce**
See OpenQA tests linked below.

**Expected behavior**
OpenQA versioning tests pass.

**OS version:**
Fedora IoT 41
Fedora IoT 43

**Additional context**
Fedora IoT 41 Tests: https://openqa.fedoraproject.org/tests/3340965#step/os_release/1

- `VERSION should be 41.20250328.0 (IoT Edition) and is 41.20250327.0 (IoT Edition)`

Fedora IoT 43 Tests: https://openqa.fedoraproject.org/tests/3340868#step/os_release/33

- `VERSION should be 43.20250328.0 (IoT Edition Prerelease) and is 43.20250327.0 (IoT Edition Prerelease)`

pcdubs, 2025-03-28:
The compose host was updated, upgrading pungi to `4.8.0-3.fc41` , which has this bug. 

The Pungi bug was fixed, but we don't have an F41 build yet, downgraded back to `pungi-4.7.0-4.1.fc41.noarch`

pcdubs, 2025-03-28:
Fixed with the downgrade - https://openqa.fedoraproject.org/tests/3341861#step/os_release/2 

Also pinged the maintainer about getting an F41 build.