fedora-iot/iot-distro #88: Consider dropping `screen` from the IoT package list

miabbott, 2025-04-01:
Fedora IoT includes two terminal multiplexers, `tmux` and `screen`. This feels duplicative.

While the installed size of `screen` is small, it is less actively developed and is missing more modern features like mouse support and status bar customization. Whereas `tmux` is more actively developed and has those modern features.

I'm not aware of any functionality in `screen` that cannot be replaced by `tmux`.

(While Fedora can make decisions independently of RHEL, it is worth noting that [RHEL deprecated `screen` in RHEL 7.6](libutempter) and it was removed in RHEL 8)

nullr0ute, 2025-04-15:
One of the reasons it's there is because tmux doesn't support serial consoles which is often used for SoCs with multiple cores and other such things that need to be debuged over similar interfaces

miabbott, 2025-05-06:
I understand the need for connecting to serial consoles for debugging, but wouldn't that typically happen from a workstation running your development tools rather than from the SoC itself?

I'm imagining trying to bring up an ARM platform from a Fedora Workstation system that is connected via serial to a dev board.

I wouldn't imagine that you would be running IoT on your workstation in that scenario.

nullr0ute, 2025-05-06:
> often used for SoCs with multiple cores and other such things

This bit is the important bit, a LOT of devices have multiple cores running different OSes such as a RTOS on a microcontroller (eg NV Jetson, Intel Elkhart Lake etc)

miabbott, 2025-05-06:
OK, I understand now.

closed 2025-05-06 by miabbott

LorbusChris, 2025-05-13:
Screen is trending on HN today: [Multiple Security Issues in GNU Screen](https://news.ycombinator.com/item?id=43971716) -> https://www.openwall.com/lists/oss-security/2025/05/12/1

Fedora doesn't seem to be affected, but maybe it's time to find an alternative anyway?
https://github.com/tio/tio is mentioned multiple times on the HN thread.