fedora-iot/iot-distro #92: Fedora IoT wan't upgrade due to Lua scripts

Programmierschmied, 2025-04-15:
**Describe the bug**
I just installed Fedora IoT with ISO and installed and uninstalled bind and bind-utils to it. After this I wanted to upgrade per rpm-ostree and he shows error.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. install via ISO
2. install bind and bind-utils and ddclient
3. uninstall bind and bind-utils
4. try to upgrade via rpm-ostree

following error:
error: Executing %transfiletriggerin for filesystem: Package 'filesystem' has (currently) unsupported <lua> script in '%transfiletriggerin'

**Expected behavior**
A clear and concise description of what you expected to happen.
upgrade to latest version of fedora IoT

**OS version:**
fedora IoT 41

```
rpm-ostree status -b
State: idle
BootedDeployment:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 41.20241027.0 (2024-10-27T21:40:59Z)
               BaseCommit: bdd952cd0d5f980407644852dbc6ccce7909e61af3f90056860a1ce0501a01f1
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
          LayeredPackages: ddclient
```

Programmierschmied, 2025-04-16:
fixed by removing ddclient...

pcdubs, 2025-04-16:
I reproduced this issue as well, I was able to upgrade to the last F41 compose before we switched F42 over to `stable`

```
rpm-ostree deploy Fedora-IoT-41-20250410.1
```

And from there an `rpm-ostree upgrade` to Fedora 42 worked ok with `ddclient` installed

```
[root@localhost ~]# rpm-ostree status 
State: idle
Deployments:
● fedora-iot:fedora/stable/aarch64/iot
                  Version: 42.20250414.0 (2025-04-14T17:18:58Z)
               BaseCommit: abbf54c00c1095967346796c41254aef00d9c76e0c36553bddb78cd139889e9e
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944
          LayeredPackages: ddclient

  fedora-iot:fedora/stable/aarch64/iot
                  Version: 41.20250410.1 (2025-04-10T20:44:10Z)
               BaseCommit: f19285331ca1a08a4f4ed2efff389a70694049416e8c666f08ece7c934019960
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1
          LayeredPackages: ddclient
```

say-paul, 2026-10-12:
Closing it now due to inactivity
Feel free to reopen the issue if you are facing same  for the latest IoT releases.

closed 2026-10-12 by say-paul