fedora-iot/iot-distro #93: grub2-mkconfig fails with error on composefs path

Procsiab, 2025-04-21:
**Describe the bug**
Whenever a change to the GRUB2 configuration is attempted, both manually through `grub2-mkconfig` or procedurally through `rpm-ostree` deployment manipulation, the operation fails with error on `grub2-mkconfig`.

**To Reproduce**
1. Deploy the OS version `41.20250318.0` as the only deployment
2. Deploy the OS version `42.20250414.0` afterwards
3. Run one of the following test cases:
  - `sudo grub2-mkconfig`
  - `sudo rpm-ostree cleanup --rollback`

**Expected behavior**
The test cases provided above should complete without errors, as in the first deployment referred in the step (1)

**OS version:**

```
State: idle
BootedDeployment:
● fedora-iot:fedora/stable/x86_64/iot
                  Version: 42.20250414.0 (2025-04-14T16:59:59Z)
               BaseCommit: 1302f1e093f1f8a64ba46ef8c6cbb1e7c932ea321522532822155d161aff1f51
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944
          LayeredPackages: bzip2 cockpit-networkmanager cockpit-ostree cockpit-podman cockpit-storaged cockpit-system unzip
```

**Additional context**
If running the fist test case, the error becomes more prominent with the following message in the output of `grub2-mkconfig`:

```
# grub2-mkconfig
/usr/bin/grub2-probe: error: failed to get canonical path of `composefs'.
```

Thanks to anybody who read this for his/her time spent on looking and thinking at this.

miabbott, 2025-09-15:
See https://github.com/fedora-iot/iot-distro/issues/122

There's a workaround there you can try.

Why are you running `grub2-mkconfig` by hand? That's generally not recommended for ostree systems.

Procsiab, 2025-09-15:
Regarding running manually the `mkconfig`, I used it as a test case to replicate the issue quickly, but I got the same behaviour trying to run the `rpm-ostree` command I posted before.

In the mean time, the proposed workaround is what I ended up using by running the command `rpm-ostree kargs --editor` and appending at the end of the line `ostree.prepare-root.composefs=0`, then finally rebooting.

closed 2025-09-15 by Procsiab