fedora-iot/iot-distro #95: Failed to pull from repository: No such branch 'fedora/41/x86_64/iot'

bam80, 2025-04-29:
**Describe the bug**
I'm trying to install F41 IoT from Kickstart.

Earlier this command worked for me:
```
ostreesetup --nogpg --osname=fedora-iot --remote=fedora-iot --url=https://kojipkgs.fedoraproject.org/compose/iot/repo/ --ref=fedora/stable/x86_64/iot
```
But now it installs F42 instead.

So I changed the line to point to 41 exactly:
```
ostreesetup --nogpg --osname=fedora-iot --remote=fedora-iot --url=https://kojipkgs.fedoraproject.org/compose/iot/repo/ --ref=fedora/41/x86_64/iot
```
and now get en error:
```
Starting pull of fedora/41/x86_64/iot from fedora-iot
================================================================================
================================================================================
Error

The following error occurred while installing the payload. This is a fatal error
and installation will be aborted.

Failed to pull from repository: g-io-error-quark: No such branch
'fedora/41/x86_64/iot' in repository summary (0)
```

**OS version:**

```
Fedora IoT 41
```

miabbott, 2025-04-29:
Fedora IoT does not use versioned branches; it uses a rolling release style with  `rawhide`, `devel`, and `stable` branches.

```
$ ostree remote list -u
fedora  https://ostree.fedoraproject.org
iot     https://ostree.fedoraproject.org/iot
$ ostree remote refs iot
iot:fedora/devel/aarch64/iot
iot:fedora/devel/x86_64/iot
iot:fedora/rawhide/aarch64/iot
iot:fedora/rawhide/x86_64/iot
iot:fedora/stable/aarch64/iot
iot:fedora/stable/x86_64/iot
```

I would also not recommend using `https://kojipkgs.fedoraproject.org/compose/iot/repo/` as that points directly at the Fedora infrastructure and doesn't take advantage of the CDN in front of the infra.

The complete ostree remote config, looks like:

```
$ cat /etc/ostree/remotes.d/fedora-iot.conf 
[remote "fedora-iot"]
url=https://ostree.fedoraproject.org/iot
gpg-verify=true
gpgkeypath=/etc/pki/rpm-gpg/
contenturl=mirrorlist=https://ostree.fedoraproject.org/iot/mirrorlist

```

bam80, 2025-04-29:
So the complete kickstart command would look like:
```
ostreesetup --nogpg --osname=fedora-iot --remote=fedora-iot --url=https://ostree.fedoraproject.org/iot --ref=fedora/stable/x86_64/iot
```
?

bam80, 2025-04-29:
> The complete ostree remote config, looks like:
> 
> ```
> $ cat /etc/ostree/remotes.d/fedora-iot.conf 
> [remote "fedora-iot"]
> url=https://ostree.fedoraproject.org/iot
> gpg-verify=true
> gpgkeypath=/etc/pki/rpm-gpg/
> contenturl=mirrorlist=https://ostree.fedoraproject.org/iot/mirrorlist
> ```

@miabbott with `ostreesetup` above got an error:
```
Failed to pull from repository: g-io-error-quark: While fetching https://ostree.
fedoraproject.org/iot/objects/f2/a7a92c7363822c198a3148af9247f94f8c3305ba3f8778a
86836c253cc1ed0.commit: Server returned HTTP 404 (1)
```

miabbott, 2025-04-29:
Hmm, I wonder if the available content on `https://ostree.fedoraproject.org/iot/` has changed...I thought you could still get the commits and objects from there...

But since it is failing, we'll have to use the URL returned from the mirrorlist:

```
# curl -L https://ostree.fedoraproject.org/iot/mirrorlist
https://d2ju0wfl996cmc.cloudfront.net/
```

If you use `ostreesetup --nogpg --osname=fedora-iot --remote=fedora-iot --url=https://d2ju0wfl996cmc.cloudfront.net/ --ref=fedora/stable/x86_64/iot` in your kickstart, that should work.

bam80, 2025-04-29:
> But since it is failing, we'll have to use the URL returned from the mirrorlist:

Where should I report about this one?
I need a stable URL for my ks to commit.

nullr0ute, 2025-04-30:
Doesn't kickstart support redirects?

bam80, 2025-04-30:
I don't know. If we absolutely sure the URL is right, I can report to them.

bam80, 2025-05-13:
> Doesn't kickstart support redirects?

Asked:
https://github.com/rhinstaller/anaconda/discussions/6408