fedora-iot/iot-distro #97: regression: Fedora IoT 42 ostree ISO Anaconda installer doesn't run %post kickstart sections

bam80, 2025-05-06:
This is regression.

With the very same kickstart, Fedora IoT 41 does run %post sections:
```
# grep -irI %post /tmp/
/tmp/anaconda.log:15:59:53,292 INF kickstart.script: Running kickstart %%post script(s)
/tmp/anaconda.log:15:59:57,469 INF kickstart.script: All kickstart %%post script(s) have been run
```
I do not see any attempts on Fedora IoT 42. %pre section runs, though.

**Steps to Reproduce:**
1. Boot Fedora IoT 42 ostree ISO with any custom ks containing %post section.
2. Let system autoinstall.
3. Without reboot, from the installer console, run:
`# grep -irI %post /tmp/`

**Actual Results:**
No relevant "%post" lines found anywhere in /tmp/* files.

**Expected Results:**
Something like this:
```
# grep -irI %post /tmp/
/tmp/anaconda.log:15:59:53,292 INF kickstart.script: Running kickstart %%post script(s)
/tmp/anaconda.log:15:59:57,469 INF kickstart.script: All kickstart %%post script(s) have been run
```

bam80, 2025-05-13:
Anybody?

nullr0ute, 2025-05-29:
Sounds more like an anaconda bug, we don't disable anything specific for the installer.

nullr0ute, 2025-05-29:
> ostreesetup --nogpg --osname=fedora-iot --remote=fedora-iot --url=https://kojipkgs.fedoraproject.org/compose/iot/repo/ --ref=fedora/latest/x86_64/iot

Updates are pushed by default so you shouldn't have to change anything to pull latest changes, that doesn't deal with the anaconda post install issues

nullr0ute, 2025-05-29:
Thanks for the confirmation

supakeen, 2025-07-18:
This bug is likely the same bug as https://github.com/osbuild/bootc-image-builder/issues/968

supakeen, 2025-07-18:
@bam80 Nothing is needed on the IoT side; this will make its way into the appropriate places automatically. Since IoT is likely going to respin the ISO that's up for download due to another (unrelated to this) [bug](https://bugzilla.redhat.com/show_bug.cgi?id=2381864) let's make sure that that's done after the above PR is shipped as well but I'll coordinate that with @pcdubs.

If you want a new ISO earlier you'd have to wait for the next `image-builder` release to build one locally.

pcdubs, 2025-07-25:
@bam80 should be fixed in https://kojipkgs.fedoraproject.org/compose/iot/Fedora-IoT-42-20250724.1/compose/IoT/

pcdubs, 2025-07-30:
@bam80 Thanks for confirming. I'll close this one as fixed.

closed 2025-07-30 by pcdubs