fedora-iot/iot-distro #101: grub not show menu

MrVan, 2025-05-15:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
After install Fedora-IoT-ostree-42-20250414.0.aarch64.iso to eMMC, grub reports failure and no menu

```
Hit any key to stop autoboot:  0
Working FDT set to 43000000
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
adv7535_mipi2hdmi adv7535@3d: Can't find cec device id=0x3c
fail to probe panel device adv7535@3d
mxs_video lcdif@32e00000: failed to get any video link display timings
  0  efi_mgr      ready   (none)       0  <NULL>
** Booting bootflow '<NULL>' with efi_mgr
Booting: Fedora
Remove /soc@0/bus@32c00000/usb@32e40000:extcon
ethernet@30be0000 Waiting for PHY auto negotiation to complete......................................... TIMEOUT !
Could not initialize PHY ethernet@30be0000
error: ../../grub-core/commands/bli.c:123:unable to determine partition UUID of boot device.

                                                                  GRUB version 2.12

   Minimal BASH-like line editing is supported. For the first word, TAB lists possible command completions. Anywhere else TAB lists possible
   device or file completions. To enable less(1)-like paging, "set pager=1".

grub>

```

**To Reproduce**
1. Install the iso to eMMC storage on i.MX8MM-EVK
2. Then boot i.MX8MM-EVK

**Expected behavior**
grub menu shows up.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**

```
Fedora-IoT-ostree-42-20250414.0.aarch64.iso
```

nullr0ute, 2025-07-28:
@pcdubs can you take a look?

MrVan, 2025-09-16:
Is there any update on this?