fedora-iot/iot-distro #102: Failed to start ostree-prepare-root.service

MrVan, 2025-05-15:
**Describe the bug**
Failed to start ostree-prepare-root.service
Need set "ostree.prepare-root.composefs=0" to bypass the step.
```
× ostree-prepare-root.service - OSTree Prepare OS/
     Loaded: loaded (/usr/lib/systemd/system/ostree-prepare-root.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2025-02-16 00:00:10 UTC; 5min ago
 Invocation: df37134ea054468c81d61980eed1a28f
       Docs: man:ostree(1)
    Process: 624 ExecStart=/usr/lib/ostree/ostree-prepare-root /sysroot (code=exited, status=1/FAILURE)
   Main PID: 624 (code=exited, status=1/FAILURE)
   Mem peak: 1.4M
        CPU: 40ms

Feb 16 00:00:10 localhost systemd[1]: Starting ostree-prepare-root.service - OSTree Prepare OS/...
Feb 16 00:00:10 localhost ostree-prepare-root[624]: Resolved OSTree target to: /sysroot/ostree/deploy/fedora-iot/deploy/abbf54c00c1095967346796c41254aef00d9c76e0c36553bddb78cd139889e9e.0
Feb 16 00:00:10 localhost ostree-prepare-root[624]: sysroot.readonly configuration value: 1 (fs writable: 1)
Feb 16 00:00:10 localhost ostree-prepare-root[624]: ostree-prepare-root: composefs: failed to mount: No such device
Feb 16 00:00:10 localhost systemd[1]: ostree-prepare-root.service: Main process exited, code=exited, status=1/FAILURE
Feb 16 00:00:10 localhost systemd[1]: ostree-prepare-root.service: Failed with result 'exit-code'.
Feb 16 00:00:10 localhost systemd[1]: Failed to start ostree-prepare-root.service - OSTree Prepare OS/.
Feb 16 00:00:10 localhost systemd[1]: ostree-prepare-root.service: Triggering OnFailure= dependencies.
```

**To Reproduce**
1. dd if=Fedora-IoT-raw-42-20250414.0.aarch64.raw of=/dev/sdb bs=128M conv=sync status=progress
2. Plug the storage image into i.MX8MM-EVK SDCARD slot, and boot

**Expected behavior**
raw image boots up successfully

**Screenshots**

**OS version:**

```
Fedora-IoT-raw-42-20250414.0.aarch64.raw
```

**Additional context**
Add any other context about the problem here.

pcdubs, 2025-05-22:
Thanks for the report. 

Were there any other changes made to the system?

MrVan, 2025-05-26:
@pcdubs  No other changes. I just put "ostree.prepare-root.composefs=0" in the grub menu linux entry, then it boots up into user terminal