fedora-iot/iot-distro #105: serial tty has too verbose dmesg

bam80, 2025-06-10:
**Describe the bug**
On newly installed system, if user logins via serial TTY (e.g. in VM), the terminal is polluted by debug log messages immediately after the login.

**To Reproduce**
1. Install the system to VM with serial TTY console
2. Boot it and login

**Expected behavior**
A shell prompt should appear.

**Actual behavior**
The full screen is polluted with debug messages immediately, before the prompt even appears.
 
**OS version:**

```
● fedora-iot:fedora/stable/x86_64/iot                                                                    
Version: 41.20241027.0 (2024-10-27T21:40:59Z)                                          
Commit: bdd952cd0d5f980407644852dbc6ccce7909e61af3f90056860a1ce0501a01f1              
```

**Additional context**
So the console log level is set to maximum by default: _debug-level messages_ (`7`):
```console
$ cat /proc/sys/kernel/printk
7       4       1       7
```
For comparison, on Fedora SB and all the other distros I faced, the default log level is _err_ (`3`): 
```console
$ cat /proc/sys/kernel/printk
3	4	1	7
```
Sorry I don't know where the default log level comes from, so can't suggest a patch.

bam80, 2025-07-02:
@nullr0ute I need more detailed instructions what I'm supposed to do, preferable in the form of docs.

pcdubs, 2025-07-02:
If you add `quiet` to the kargs, it will align with the other editions (which also have quiet). 

```
ssh root@fitlet2
Boot Status is GREEN - Health Check SUCCESS
Last login: Wed Jul  2 16:47:20 2025 from 192.168.1.11
[root@localhost ~]# cat /proc/sys/kernel/printk
3	4	1	7
```

You can use `rpm-ostree kargs --editor` to add it.

bam80, 2025-07-02:
Thank you, but:
- I don't want to use `quiet` (what exactly it does, BTW?)
- other editions do not require it

nullr0ute, 2025-07-02:
> [@nullr0ute](https://github.com/nullr0ute) I need more detailed instructions what I'm supposed to do, preferable in the form of docs.

As per [IoT docs for layering packages](https://docs.fedoraproject.org/en-US/iot/add-layered/). So that would be `sudo rpm-ostree install audit`

pcdubs, 2025-07-02:
> * other editions do not require it

The include it out of the box.

bam80, 2025-07-02:
> As per [IoT docs for layering packages](https://docs.fedoraproject.org/en-US/iot/add-layered/). So that would be sudo rpm-ostree install audit

I'm aware how to install layered packages, what I mean is it should be documented `audit` package is required to have normal tty behavior, if IoT doesn't install it by default for some reason.

francoism90, 2025-07-03:
I also find IoT to verbose.

@pcdubs I didn't know about the `quiet` flag workaround, thought it only did make the bootup less verbose.

pcdubs, 2025-07-03:
> > I also find IoT to verbose.
> 
> Can you try the audit layer and confirm if it solves your expectations. We don't currently have it in base because of dependencies, but like all things it's a balance

Adding `audit` does quiet things down, it pulls in a couple of deps:

```
Added:
  audit-4.0.5-2.fc42.x86_64
  audit-rules-4.0.5-2.fc42.x86_64
  initscripts-service-10.26-3.fc42.noarch
```

bam80, 2025-07-12:
As a workaround, I currently have following in my Kickstart file.
But that costed me quite a lot of research, which should not be required from the user:
```sh
( . /etc/os-release; [ $VARIANT_ID = iot ] && echo 'kernel.printk = 4 4 1 7' > /etc/sysctl.d/20-quiet-printk.conf )
```