fedora-iot/iot-distro #110: Installer does not run on PC

francoism90, 2025-06-26:
This issue tracker is intended only for IoT specific issues. Please try to reproduce the issue on a relevant Fedora release to determine if the issue is specific to IoT or a general issue in Fedora. If is a general issue in Fedora, please report it in [Red Hat Bugzilla](https://bugzilla.redhat.com/) (see [How to file a bug](https://docs.fedoraproject.org/en-US/quick-docs/howto-file-a-bug/)) or in an upstream project and not in this issue tracker.

**Describe the bug**
I get a gray screen, installer doesn't show up. It does work fine in a VM.

**To Reproduce**
Please describe the steps needed to reproduce the bug:
1. Create bootable USB using Fedora USB Writer
2. Boot USB
3. No setup/installer, only a grey background and cursor
 
**Expected behavior**
Installer starting

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OS version:**

Latest iso on homepage

```
Please replace this line with output of `rpm-ostree status -b`
```

**Additional context**
Add any other context about the problem here.

nullr0ute, 2025-06-27:
What do you mean by PC? Can you give more details, at what point does the screen go grey, what output do you get. What sort of PC is it? If it's an older device is it in UEFI or BIOS mode?

francoism90, 2025-07-04:
@pcdubs Confirm, also the fallback setup mode doesn't work. I think this also adds nomodeset, but also adding it myself doesn't change anything.

closed 2025-07-21 by pcdubs