fedora-iot/iot-distro #122: `failed to finalize previous deployment` when trying to upgrade from 42.20250902.0

miabbott, 2025-09-15:
I have a Raspberry Pi 3B that has Fedora IoT installed. 

It had sat dormant (powered off) for a few months and I finally got around to powering it on and seeing where it was at. 

I don't recall the exact version I found it on, but I suspect it was in the Fedora 41 era.

I was able to get it upgraded to 42.20250902.0. However, when I tried to upgrade to the latest (42.20250915.0), I was met with an error:

```
[core@fedora-iot-pi ~]$ rpm-ostree status                                                                                                                                                                                                                                                                         
State: idle                                                                                                                                                                                                                                                                                                       
Warning: failed to finalize previous deployment                                                                                                                                                                                                                                                                   
         error: Bootloader write config: grub2-mkconfig: Child process exited with code 1                                                                                                                                                                                                                         
         check `journalctl -b -1 -u ostree-finalize-staged.service`                                                                                                                                                                                                                                               
AutomaticUpdates: stage; rpm-ostreed-automatic.timer: last run 51min ago                                                                                                                                                                                                                                          
Deployments:                                                                                                                                                                                                                                                                                                      
  fedora-iot:fedora/stable/aarch64/iot                                                                                                                                                                                                                                                                            
                  Version: 42.20250915.0 (2025-09-15T11:09:03Z)                                                                                                                                                                                                                                                   
                   Commit: 7f8d9a96f8a792bac4e4bf20428d72a4d93a75f23479d50020cce85323a4e995                                                                                                                                                                                                                       
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944                                                                                                                                                                                                                            
                     Diff: 38 upgraded                                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                                                  
● fedora-iot:fedora/stable/aarch64/iot                                                                                                                                                                                                                                                                            
                  Version: 42.20250902.0 (2025-09-02T11:08:31Z)                                                                                                                                                                                                                                                   
                   Commit: c935cf53197eca66511aa86b8ab20c874918fdeb5f5db7372bde1e71f28b68a3                                                                                                                                                                                                                       
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                                                                  
  fedora-iot:fedora/stable/aarch64/iot                                                                                                                                                                                                                                                                            
                  Version: 41.20241117.0 (2024-11-17T12:08:07Z)                                                                                                                                                                                                                                                   
                   Commit: 0f925474d6048d252da2cfd8eaef587492d5a3300d8187dd4f86fdeafd4e85dd                                                                                                                                                                                                                       
             GPGSignature: Valid signature by 466CF2D8B60BC3057AA9453ED0622462E99D6AD1                                                                                                                                                                                                                            
[core@fedora-iot-pi ~]$ sudo journalctl -b -1 -u ostree-finalize-staged.service                                                                                                                                                                                                                                   
Sep 15 16:29:38 fedora-iot-pi systemd[1]: Finished ostree-finalize-staged.service - OSTree Finalize Staged Deployment.                                                                                                                                                                                            
Sep 15 17:46:01 fedora-iot-pi systemd[1]: Stopping ostree-finalize-staged.service - OSTree Finalize Staged Deployment...                                                                                                                                                                                          
Sep 15 17:46:02 fedora-iot-pi ostree[2051]: Finalizing staged deployment                                                                                                                                                                                                                                          
[... cut ...]
> You should add `ostree.prepare-root.composefs=0` at the end of the line that starts with `linux ($root)/ostree/...` and ends with `rw`. You can add it after the `rw`, making sure to add a space.

I powered on my Pi, edited the current boot entry with that workaround, was able to `rpm-ostree upgrade` to latest, then rebooted into the new version.

My hunch is this has something to do with jumping a lot of versions on upgrade. As far as I can tell from the journal on the device, this device started back in F39, upgraded to F41, and then to F42.

Since the issue tracker isn't overflowing with reports like this, I am hoping this is isolated.

pcdubs, 2025-09-17:
Trying to reproduce I deployed a VM with `Fedora-IoT-raw-41-20241027.0.aarch64.raw.xz`

Then upgraded to `rpm-ostree deploy 42.20250902.0`. From there I was able to upgrade to `rpm-ostree deploy 42.20250915.0` without issue:

```
rpm-ostree status 
State: idle
Deployments:
● fedora-iot:fedora/stable/aarch64/iot
                  Version: 42.20250915.0 (2025-09-15T11:09:03Z)
                   Commit: 7f8d9a96f8a792bac4e4bf20428d72a4d93a75f23479d50020cce85323a4e995
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944

  fedora-iot:fedora/stable/aarch64/iot
                  Version: 42.20250902.0 (2025-09-02T11:08:31Z)
                   Commit: c935cf53197eca66511aa86b8ab20c874918fdeb5f5db7372bde1e71f28b68a3
             GPGSignature: Valid signature by B0F4950458F69E1150C6C5EDC8AC4916105EF944

```

pcdubs, 2025-09-18:
Fwiw, I started with F38 and upgraded (rpm-ostree deploy $version)  through to F42 without issue, composes used:
```
Version: 38.20231103.0
Version: 39.20240417.0
Version: 40.20240424.0
Version: 41.20250411.0
Version: 42.20250917.0

```